jgyates/genmon #382: NSURLErrorDomain

jjmacneil, 2020-05-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Login page

### Actual Behavior

NSURLErrorDomain error message

### Steps to Reproduce (including precondition)

Trying to visit the management webpage

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model:
- Generator Registers: Submit via the About page
- Genmon Version: After upgrade to 1.14.08

jgyates, 2020-05-20:
I am going to need more info before I can help.

Is this a new installation? Have you ever been able to access the web interface?

What is the IP address of you pi and what address are you using in your browser? Can you ping your raspberry pi?

jjmacneil, 2020-05-21:
Hello,

It’s not a new install. I’ve been running GenMon since February 2019 and have updated many times since the initial install with no trouble. I was running 1.14.06 with no trouble until after upgrading to 1.14.08

Yes, I can ping the Pi and the address is 192.168.100.210

I’ve been logging in at https://192.168.100.210 since the initial install with no trouble.

I rebooted the Pi a few times with no luck

jjmacneil, 2020-05-21:
I can also connect to the Pi via SSH without a problem so the issue appears to be the GenMon application itself preventing access to the login page for some reason.

jgyates, 2020-05-21:
if you can login via ssh into the pi then do this:

    cd genmon 
    ./genmonmaint.sh -u
    ./startgenmon.sh restart

This will upgrade to the latest and restart the software. There was a bug introduced in 1.14.08 that could cause this. The latest version is V1.14.10.

jgyates, 2020-05-21:
fyi, you will need to type "y" to confirm the upgrade in step 2

jjmacneil, 2020-05-21:
Thank you! I followed the steps, it updated, restarted and GenMon is now working. Thanks again for the quick response. I’m so frustrated to hear that Generac is encrypting their controllers to lock out third party access as the GenMon app is so much better than what Generac can provide with their mobile link application.

jgyates, 2020-05-21:
I am glad it is working. Sorry about the bug. I am going to close this for now. If you have any other issues on this topic you can post them here.

closed 2020-05-21 by jjmacneil