jgyates/genmon #383: Generator does not have the molex port?

skipfire, 2020-05-22:
Hoping for confirmation that I may have a problem.  I got my serial loopback test working and we have dry weather so I opened the generator looking for the port to go plug into, and did not find an empty one.  I did find an 8-pin molex plugged in, next to that is a slot for an 8-pin but just through holes, no pins and no connector.  The 8-pin that is there has 4 wires in use (1,2,5,6 I think), but they disappear into another space.  The empty space in the picture was covered by a sticker.

1. Can anyone confirm that I found where there should be a port?
2. Might that be going to the wifi?  If so, perhaps can use an 8-pin extension and splice into the the 2 pins I need?
3. If I'm not using WiFi, can I just unplug it completely or will it cause other issues?
4. I really don't want to solder pins onto this board.

So far I have done nothing other than remove the front and side panels of the generator enclosure, so worst case I very sadly give up and use the absolutely terrible app that they do have.

Evolution 2.0 controller
Model: G0070432
Prod Date: 20190318
Installed: 20190425

[image]

liltux, 2020-05-22:
[image]
@skipfire  circled the molex or minifit-jr connector you will need to utilize.  currently your wifi module is plugged into that port.  if you are not using the wifi module, just unplug.  warning, as of right now the latest firmware 1.10 on evo 2 is causing communication issues with genmon.  be sure your firmware is not greater than 1.09.  if you are choosing to use the RPiZero(w), do not power from the evo2 board.  when the wifi module is disconnected from the board for a predetermined time (about 5-10 min.) the board will see the module as not present and will turn off the 12vdc output on that connector.  coms will still be available and will need to power your device from the generator battery via a 12v to 5v adapter of some sort.

liltux, 2020-05-22:
Also to note: in that picture the blue and white are the RS485 to the wifi module.  The Black and Red are DC-(black) and DC+(red).

skipfire, 2020-05-22:
@liltux thanks, from the documentation I was expecting an unoccupied port and saw the empty one with no pins.  I will go and pull and pray.

closed 2020-05-22 by skipfire