jgyates/genmon #384: The Generator status (values) are not displayed

mxplea10, 2020-05-23:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

 

### Actual Behavior

After more than two months of working error free the generator's value are no longer showing in the genmon status.  When the cursor is over the  exclamation point,  GENMON System Warning - System Initializing is displayed.
[image]
[image]
[image]

### Steps to Reproduce (including precondition)

I revalidated the RS232 cable by testing the loopback of the transmit and receive pins (7 and 8) on the molex connector.  I, also, issue the commands _bash ./genmonmaint.sh -i_ and _bash ./genmonmaint.sh -r_ since I noticed the error after the previous upgrade. I verified that the transmit, receive and ground pins were all secured in the molex connector inside the generator and  inside the network cabinet on the Raspberry Pi side.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: V1.14.10

jgyates, 2020-05-23:
Do you have an Evolution 2.0 controller? If so then this thread speaks to the issue you are likely having:  https://github.com/jgyates/genmon/issues/381

On the Monitor page if you are showing modbus exceptions then this is your issue.

mxplea10, 2020-05-23:
Yes, I have the Evolution 2.0 controller.  I'll check #381.  Thanks.

mxplea10, 2020-05-23:
The firmware version is 1.10.059.

jgyates, 2020-05-23:
I am going to close this issue as it is a duplicate to #381

closed 2020-05-23 by jgyates