jgyates/genmon #386: Source for Battery Charger connectors

jwtaylor310, 2020-05-26:
Does anyone know of a source for the two-wire locking connectors used to connect the battery charger to the Evolution 2.0 controller?  I periodically have to unplug this cable (and remove the controller fuse) to reset serial communications to my Pi.  I want to extend the existing cable to a switch to allow me to disconnect the circuit without the need to remove the enclosure's end panel to get at the connector.  If I can find a source for the connectors I will be able to do this without cutting into the factory wiring.  Thanks!

[image]

liltux, 2020-05-26:
@jwtaylor310 , I do not have a good source for that connector, but I do have information.  The 110/120VAC supply that goes through that connector to the controller is supplied throught he T1 connector in the customer connections compartment of the generator.  This wire is terminated at the T1 connection at the transfer switch.  Therefore, one could remove that wire, add a butt-splice or wire-nut, connect that wire to one side of a switch.  To the other side of that switch run a wire back to the T1 connection that was removed.   
[image]
Switch can be installed anywhere in the T1 wire between the Generator and Transfer switch to do the job you are wanting to do.

jwtaylor310, 2020-05-27:
liltux - Thank you for the suggestion.  I did not realize that the circuit went all the way to the transfer switch.  That offers several opportunities for 'intercepting' the feed without cutting up the factory wiring.  i'll trace out the wiring and see what I can do.

skipfire, 2020-05-28:
@jwtaylor310  I'm wiring in a screw terminal USB adapter from the 8-pin that supplies serial, and then using a USB switch cable that will be positioned outside the screw-only access area so I can power cycle the Pi easily if necessary.  Plenty of variation available on both parts.  But this will have me only with 1 connection to the 8-pin molex and nothing else.  Also have a serial hat on order that will fit inside the HighPi case so I will have a very clean install.  On my generac with Evo2 controller I have a space in front of the controller where there is room for the cables to run through and for the switch to sit.

https://www.amazon.com/gp/product/B07JMBSSPW/ref=ppx_yo_dt_b_asin_image_o07_s01?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07T9BRNHW/ref=ppx_yo_dt_b_asin_image_o07_s00?ie=UTF8&psc=1

jwtaylor310, 2020-05-29:
liltux: I've inspected my generator's factory wiring and you are correct.  I can reset the controller by   pulling the controller's fuse and then momentarily opening the T1 circuit either at the transfer switch or in the generator housing.  I have opted to install a Normally Closed (NC) push button on the inside front panel of the transfer switch.  The outer cover panel will protect the switch from the weather and the switch won't be subject to engine heat and vibration as it would be if mounted in the generator housing.  I plan to lift the field-installed T1 wire at the 'Customer Control Connections' terminal block and reroute it to one side of the NC switch.  Then I will run a new wire from the other side of the NC switch to the terminal block.  Thank you for your suggestion!

liltux, 2020-05-29:
@jwtaylor310 no problem.

jwtaylor310, 2020-05-29:
skipfire: As you figured out, my need is be able to easily interrupt the power to the controller.  My Pi is located in my garage about 75 feet from the generator.  However, anytime that serial communications between the Pi and the generator is interrupted the generator's controller shuts down its serial port.  In my system that happens if I turn off the Pi for any reason and, occasionally, when Generac pushes an update to the controller.  Rebooting the Pi doesn't fix the problem since it is the generator's port that shuts down.  The only solution that I have found is to pull the controller's fuse and then disconnect the battery charger circuit's "T1" line.  This thread is about finding a way to open the T1 line without having to open the generator and remove the front and side panels to get at the T1 connector inside the control wiring bay.  Good luck with your installation.  Genmon is a great product and jgyates provides outstanding support for it!

closed 2020-05-29 by jwtaylor310