jgyates/genmon #387: Genmon web interface becomes non-responsive

smeisner, 2020-05-26:
Running the latest version of Genmon and am seeing a weird issue lately. I cannot tell if this is an issue with Genmon or not yet. Any help debugging this issue further would be appreciated.

I am using a Pi Zero W over wifi. This setup has been running for over a year now. After starting Genmon everything works fine. But after some number of hours, the web site becomes unresponsive. If I do a `netstat -lpn` on the Pi, I get:

```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp      129      0 0.0.0.0:443             0.0.0.0:*               LISTEN      7435/python

```
Notice the receive queue is at 129. I assume this is why the web site is not responding. There are no warnings or errors in the system or Genmon log files. So again, what can I do to debug this a bit further prior to opening a more descriptive ticket?

When this situation is happening, I can still log in via ssh.

jgyates, 2020-05-26:
Have you tried a reboot? It is possible that the flask library is having an issue and just needs a restart. 

Another option would be to update the system software and restart:

     sudo apt-get update && sudo apt-get dist-upgrade
     sudo reboot

Also, check your wifi signal strength. Strange things can happen if you have a marginal signal.

smeisner, 2020-05-26:
Yes, updating (update & upgrade) via apt was the first thing I did.

And, yes, monitoring wifi was something I was doing. But I didn't see anything interesting. The signal did stay fairly strong the entire time.

jgyates, 2020-05-27:
do you get the same results if you do not use https? I am wondering if it is related to https or your certificate if you are using one.

smeisner, 2020-05-27:
I am only listening on 443, so I only test with https. The cert is from Let's Encrypt ... and again has been in use for 6 -7 months now (renewing via automated cert tool) with no problems until ~1+ month ago.

To gather further details, I have set up 3 scripts running simultaneously;
- 1 that pings my server (so across wifi) and logs the output
- 1 that logs the wifi signal strength info 
- 1 that does a `netstat -lpn | grep '443'` every 5 secs. If the output from netstat changes at all, it will cause all 3 scripts to stop, leaving the latest info from each script on the screen. Since the symptom is the recv queue backing up, I am using that to trigger the stop of all scripts.

Once this test is done, I will try re-enabling http and test that for a bit.

smeisner, 2020-05-28:
The situation did recur yesterday. The first test was roughly 4 hours after starting the test. The second time took about 8.5 hours. I've included the log files of the output from 2 of the scripts; the netstat output showing when the recv queue started backing up and the link quality log over the same time. The wifi quality (and connectivity) was fine when things started going south.

Next, I will now try using http instead of the ssl cert.

[genmon - netstat.txt](https://github.com/jgyates/genmon/files/4695616/genmon.-.netstat.txt)
[genmon - Link quality.txt](https://github.com/jgyates/genmon/files/4695618/genmon.-.Link.quality.txt)

smeisner, 2020-06-03:
Well, since my last post (May 28), before I tried the http route, I updated to 1.14.10 and waited a couple of days. I am still using the Let's Encrypt cert and https and am not seeing the problem any longer. So again, the only difference was updating genmon. I do not know what side effects it had, but the problem is no longer there.

Closing this issue.

closed 2020-06-03 by smeisner

jgyates, 2021-10-15:
@smeisner 

Thanks for the follow up! This is good info.