jgyates/genmon #388: fuel tanl level monitor

grsthegreat, 2020-05-27:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}I just added the tank utility monitor to my genmon program, and i acyivated and registered the email address on the add on page. this added an EXTERNAL TANK icon, but the leven shows empty. i reread the past posts about this, and i cant seem to find out what i did wrong.

when i go to tank utility site it shows my tank at 78%. i verified all info and log on. i do know when i added the add on to my unit, i received 9 emails from my generator.

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-05-27:
@grsthegreat 
I will need to see the log file. The file /var/log/gentankutil.log will contain any relevant errors for the tankutil add on program. You can submit your log files or post the contents of the file to this thread.

grsthegreat, 2020-05-27:
id like to, but now i cant access my generator monitor. i think i have to reboot it

grsthegreat, 2020-05-27:
ok, i rebooted genmon, i can see it using FING, but when i type in the address i get unable to connect. i tried to reboot a second time, same thing

jgyates, 2020-05-27:
can you ping the ip address of the pi?

Can you access the pi via ssh?

grsthegreat, 2020-05-27:
i can ping it in ING

whats ssh

grsthegreat, 2020-05-27:
i meant to say, i can ping it using FING program

grsthegreat, 2020-05-27:
ok, weird. i can log onto it using PuTty and using ip address. but not sure what to do past that

jgyates, 2020-05-27:
ok. ssh is what you are using to log in. PuTty is just a user interface on top of ssh.

Once you log in then type:

     /var/log/gentankutil.log

jgyates, 2020-05-27:
sorry type

      cat /var/log/gentankutil.log

grsthegreat, 2020-05-27:
all it says is error logging in TankCheckThread, retrying
[image]

grsthegreat, 2020-05-27:
Why do you think I can no longer view my generator monitor? I have tried rebooting it 2 times. It can see the pi unit, but I cant access it

jgyates, 2020-05-27:
Sorry, I thought you resolved your issue since you closed the issue. 

I am confused about the specific problem you are having. You initially asked about the tank utility add on program. Now you appear to have an issue accessing the web interface. Were you able to access the web interface earlier? If you use wifi to access the generator monitor then your problems could be caused by interference in your wifi signal or a marginal signal strength.

When you try to access the web interface, are you using the same IP address that you use to log in via ssh / PuTTY?

You tank utility log states that there is a problem logging in. Possible issues there are:

 your credentials are incorrect
the pi can not access the internet
the tank utility server is having issues.
problems with your local network

If you log into the pi you can type this:

     cd genmon
     ./startgenmon.sh restart
      python ClientInterface.py

If genmon is running, this will give you a  prompt. From the prompt type

     generator: monitor

Copy and paste the output here.

To exit the ClientInterface.py program type:

    exit

To close the ssh session type "exit" again:

     exit

grsthegreat, 2020-05-27:
ok, my genmon was working great until i activated the tank utility addon. i put in the credentials that i registered with tank utility. i was able to go to their site and i can see my tank using those credentials.

after i entered those credentials on the genmon portal, the system crashed. now i cant get into the portal anymore

by the way...i accidentally hit the close issue button.

grsthegreat, 2020-05-27:
i wwent into putty and this is what i get  

[image]

jgyates, 2020-05-27:
Looks like your /etc/genmon/genloader.conf file is corrupt. To fix this issue do this:

1) Log in via putty:
2) then type:
     
     cd genmon
     sudo cp ./conf/genloader.conf /etc/genmon
     ./startgenmon.sh restart

You will have to re-enable the tank utility add on (and any other add ons you were using) from the add on page. The rest of your settings should remain in place.

grsthegreat, 2020-05-27:
oh....your wonderful. that did it

ill try redoing the tank utility now...ill keep my fingers crossed.

your amazing

grsthegreat, 2020-05-27:
ok, with that ixed i went in and redid the external uel tank monitor. i entered tank name, username and password. it doesnt like my password. however when i go onto tankutility site, i use the same password and i can access it....very odd. 

this add n is getting info from tank utility site is it not?

also, under tank name does it want to know the name i set for monitor or the 8 digit identifier that came with unit?

grsthegreat, 2020-05-27:
ok.i deleated all the inormation and turned of the monitor. then i reinstalled it and now it works.

very odd, its the same info as before....but im not going to complain. i am just going to fix myself a stiff drink

jgyates, 2020-05-27:
I am not sure. When I wrote the tank utility app someone let me have access to their tankutility account. I don't have that access any longer. The third parameter should optional if you only have one tank. I am going to close this thread. If you still have any issue or questions on this topic feel free to post them in this thread.

closed 2020-05-27 by jgyates