jgyates/genmon #389: serial0 -> ttys0 instead of ttyAMA0

nekitti, 2020-05-28:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon talking to generator

### Actual Behavior

Not talking

### Steps to Reproduce (including precondition)

My Genmon is passing serialconfig and serialtest, but not talking to controller.  My serial port seems to be set incorrectly.  Can you tell me how to change this?  Thanks so much.  ROb

### Screenshot or Pictures relating to the problem (if possible)

and confirm that /dev/ttyAMA0 has an alias of /dev/serial0. The output should look like this:

 lrwxrwxrwx 1 root root 7 Apr 12 01:06 /dev/serial0 -> ttyAMA0
 lrwxrwxrwx 1 root root 5 Apr 12 01:06 /dev/serial1 -> ttyS0
If serial0 -> ttyS0 instead of serial0 -> ttyAMA0 then you may have attempted to enable the UART via raspi-config. This may cause interference with the console. The onboard serial port is ttyAMA0 and for compatibility reasons you should typically use the alias. In short, to user the onboard serial port with genmon you must user the alias mapped to ttyAMA0, not ttyS0.

### Your Environment

- Generator Model: 11Kw Generac
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-05-28:
@nekitti 

Take a look at this thread as I think it covers the same ground. 

https://github.com/jgyates/genmon/issues/340

Let me know if you have any questions.

closed 2020-05-29 by jgyates

nekitti, 2020-05-30:
Ok I've tried several things and am getting nowhere.  Please help.  

I should have stated in the first post that this was a functioning Genmon system.  I updated a few weeks ago and the next time I went to log on, it wasn't showing up.  I rebooted the Rasberry Pi and still nothing.  I then reformatted the SD card and reinstalled all the OS for the Pi and Genmon.  That is when I started having the problem with serial0 -> ttS0.

I changed the serial port on the Genmon software settings to Serial 1 and that didn't help.

Not knowing what to do next, I reformatted the SD and reinstalled everything.  It is passing Serialtest and serialconfig.  I can't seem to be able to check the serial.  When I type ls –l /dev/serial*  I get an error of: ls: cannot access '-l': No such file or directory

Sorry for my ignorance on this stuff.  Thanks for the help.
Rob

jgyates, 2020-05-30:
Hi @nekitti 

Since this was working I have to ask this: Do you have an Evolution 2.0 controller or and Evolution 1.0 controller? There is a know issue with an Evolution 2.0 controller. If you have an evolution 1.0 read on.

Can you describe what the genmon software is doing or not doing? I would help me understand your base problem. You mention that "I changed the serial port on the Genmon software settings to Serial 1 and that didn't help." and "it wasn't showing up" but in order to understand I will need a more descriptive details, for example I can not access the web interface or the web interface comes up but serial communications fail.

I assume you can log into the pi via ssh or puTTY since you are trying to type "ls -l /dev/serial*"

When I copy and past your command I get the same problem, but when I type it on my system it works. Try copying and pasting this:

      ls -la /dev/serial*

There appears to be a problem with the "-" in your statement. Are you using an English keyboard? You and leave off the "-l" and just type ls /dev/serial*.

Does genmon load, can you access the web interface, if yes what does the Monitor page show?

nekitti, 2020-05-30:
It's and Evolution 1 controller.  11kw air-cooled.  I can open up Genmon from other computers on my network.  Everything looks normal and I can access all the tabs. Genmon shows all the data for the Rasberry Pi P3B+, but all the data for the generator is 0.  

To setup and install the software on the Pi I have been removing the hardware from the Generator and hooking up a mouse, keyboard and monitor.  

ls: cannot access '-l': No such file or directory
This is just the way that the error showed up in terminal when I tried to check the serial designation.  I'll try just typing it in, rather than copy and paste.

I have been using Rasberry Pi Imager to install the OS on the SD card.  I have also been installing the regular version of Rasperrian.  Do you think that could be a problem?

nekitti, 2020-05-30:
Ok when I type in the command for checking the serial it works.  It doesn't like copy and paste.
Here is what I get:
pi@raspberrypi:~ $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 5 May 30 12:06 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 May 30 12:06 /dev/serial1 -> ttyAMA0
pi@raspberrypi:~ $ 

As you can see serial0 ->ttyS0 not ttyAMA0

Thanks for helping with this.

jgyates, 2020-05-30:
I had asked about the contents of the Monitor page as this would let me know the state of the serial communications. It would be helpful to see that info.

Based on your data I am assuming that you can pull up the web interface but you are not receiving data, but with the data I have that is a guess on your situation. Please try these two things:

1) Change the serial port the /dev/serial0 on the settings page, then apply the settings. Does this work?
2) Change the serial port the /dev/serial1 on the settings page, then apply the settings. Does this work?

nekitti, 2020-05-30:
Tried Serial0 and Serial1 on settings page and neither worked. Sorry about not posting screen shots.  Here they are:
[image]
[image]

jgyates, 2020-05-30:
Looks like your serial comms are not functioning. I would run through the items listed in this doc:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

This page includes testing of your pi, the RS232 converter and the cable.

nekitti, 2020-05-30:
Yes.  I already ran those test and I just ran them again to be sure.  serialtest.py passed! loopback successful and serialconfig.py -c all ok.  Power and ground good.

Here is the /var/log/myserial.log
2020-05-30 16:47:00,789 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:00,825 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:00,949 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:00,950 : Error in SerialDevice:Flush : /dev/serial0:(9, 'Bad file descriptor') : myserial.py:165
2020-05-30 16:47:00,968 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,100 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,186 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,229 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,315 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,366 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,367 : Error in SerialDevice:Flush : /dev/serial0:(9, 'Bad file descriptor') : myserial.py:165
2020-05-30 16:47:01,459 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,509 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120
2020-05-30 16:47:01,544 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120

And the /var/log/mymodbus.log
2020-05-30 16:47:14,165 : Error Receiving slave packet for register 020a
2020-05-30 16:47:14,455 : Discarding byte slave: 00
2020-05-30 16:47:14,457 : Error Receiving slave packet for register 000d
2020-05-30 16:47:14,469 : Discarding byte slave: 00
2020-05-30 16:47:14,471 : Error Receiving slave packet for register 000e
2020-05-30 16:47:14,504 : Discarding byte slave: 00
2020-05-30 16:47:14,506 : Error Receiving slave packet for register 000f
2020-05-30 16:47:14,518 : Discarding byte slave: 00
2020-05-30 16:47:14,519 : Error Receiving slave packet for register 024a
2020-05-30 16:47:14,531 : Discarding byte slave: 00
2020-05-30 16:47:14,532 : Error Receiving slave packet for register 000a
2020-05-30 16:47:14,565 : Discarding byte slave: 00
2020-05-30 16:47:14,567 : Error Receiving slave packet for register 000b
2020-05-30 16:47:14,578 : Discarding byte slave: 00
2020-05-30 16:47:14,579 : Error Receiving slave packet for register 000c

It seems like the problem is with the serial0 not setting to ttyAMA0.  Any thoughts on how to fix this? Thanks

liltux, 2020-05-30:
@nekitti can you copy the last bit of your config.txt?
@jgyates raspbian made changes for pi 4 in the config.  I had an issue once with a Pi 3 because of how they changed the outline of the file.  The serial dtoverlays maybe under the pi4 section.  When booting with pi 3 those changes are ignored but your script passes ok.  FYI

jgyates, 2020-05-30:
@nekitti , the full path of the file would be /boot/config.txt so from the console type:

     sudo cat /boot/config.txt

Then copy and paste the last 15 lines here

nekitti, 2020-05-30:
Is this enough info?
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=pi3-disable-bt
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1

jgyates, 2020-05-30:
One cause of this error "/dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)" is that some other software is trying to access the serial port. You mentioned you are running the full version of raspbian. I recommend the lite version as the gui is not needed and only takes away from resources to have it running in the background. There could be something running from the graphical interface that is trying to access the serial port. You could try reinstalling with the lite version.

liltux, 2020-05-30:
@nekitti need to move the 
dtoverlay=pi3-disable-bt 
Below the [all] section.

nekitti, 2020-05-30:
> @nekitti need to move the
> dtoverlay=pi3-disable-bt
> Below the [all] section.
> 
> And reboot, almost forgot.

Okay. How do I do this?

jgyates, 2020-05-30:
sudo nano /boot/config.txt

then make the change in the text file. Then type Ctrl+X, then y, then Enter to save and exit

liltux, 2020-05-30:
what @jgyates said, sorry I had to cook some hotdogs over a fire.

nekitti, 2020-05-30:
Okay.  I change config.txt to move the disable bt below all.  Tried it and it still wasn't talking to Generator.  I then change to /dev/serial1 on the settings page of Genmon and now I can't get genmon to show up on the web interface.  Any thoughts?  Should I try reinstalling everything with Rasberrian lite?

liltux, 2020-05-30:
Are you getting the same errors

nekitti, 2020-05-31:
Now I seem to be getting this error
2020-05-30 18:41:42,753 : Error on open serial port /dev/serial1: Could not configure port: (5, 'Input/output error')

jgyates, 2020-05-31:
when you do a loopback test. Which port passes the loopback? /dev/serial0 or /dev/serial1

Can you get the port to fail the loopback test if you remove the loopback?

nekitti, 2020-05-31:
It passes serial0  loopback test.  Yes it fails if I remove loopback jumper.
Where can I change it back to /dev/serial0 in the terminal settings?

liltux, 2020-05-31:
Edit genmon.conf
sudo nano /etc/genmon/genmon.conf
Make you change
Ctrl+x 
Restart genmon
sudo bash /home/pi/genmon/startgenmon.sh restart

jgyates, 2020-05-31:
ok. Let's change the genmon setting back to /dev/serial0

if you can not get the web interface to load then you can edit the genmon.conf file.

     sudo nano /etc/genmon/genmon.conf 

change the line:

    port = /dev/serial1

to 

    port = /dev/serial0

then type Ctrl+X, then y then enter to exit and save the file.

If you are in the genmon directory (cd genmon) then you can type:

    ./startgenmon.sh restart

to restart genmon. If your cable is connected to your controller and you still get the "/dev/serial0:device reports readiness to read but returned no data " errors  in /var/log/myserial.log then I would reinstall with the lite version. Based on this error you likely have some other program trying to use the serial port.

nekitti, 2020-05-31:
Got it set back to Serial0 and now I have web interface, but still not talking to controller.  I'll reinstall everything with Rasberrian lite tonight.  Thanks for all the help both of you.  I'll let you know how it goes.
Thanks,
Rob

nekitti, 2020-06-02:
Reinstalled everything and tried in both Rasperian and Rasperian Lite.  Got everything setup correctly.  Serial0->ttyAMA0, serial tests ok, Loop test passed.  Web interface works, but still no data from controller.  At this point I'm thinking either the controller isn't putting out data or there is a glitch in the program.  This all did seem to happen after I updated to v1.14.10, but maybe just coincidence.  My controller is Evolution 1.
[image]
[image]

jgyates, 2020-06-02:
When you are doing the loopback, did you do a loop back thru your cable? You could have a bad connection to your controller or the controller has a bad RS-232 drivers. I don't see any evidence of a program glitch. The program is behaving (timeouts) like it is transmitting but nothing is coming back in response.

nekitti, 2020-06-02:
Yep tested it at cable end. Have you ever heard of a controller working fine, but not talking to the serial port?  Thanks

skipfire, 2020-06-02:
Is it possible that the controller isn't communicating on serial like the report in issue #386?  Maybe pulling the fuse and disconnecting the battery to reset the controller?

liltux, 2020-06-02:
@nekitti that sounds like a familiar thread on here somewhere.  Might search the closed threads.  Also, have you tried to complete power cycle of the controller yet?  May just be the internal logic is locked up and needs a "reboot" on the Evo controller.

nekitti, 2020-06-02:
> @nekitti that sounds like a familiar thread on here somewhere. Might search the closed threads. Also, have you tried to complete power cycle of the controller yet? May just be the internal logic is locked up and needs a "reboot" on the Evo controller.

How do you do a complete power cycle?  I pulled the fuse and disconnected the battery, but that didn’t seem to do anything. The controller was still powered up. Thanks for the help everyone.

jgyates, 2020-06-02:
https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

nekitti, 2020-06-02:
> https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

Thanks. I didn’t know how to search the wiki. I’ll give this a try.

nekitti, 2020-06-02:
Reset/power cycled  the controller and the Genmon.  Still no communication between the two.

jgyates, 2020-06-14:
@nekitti 

here is what I recommend if you are still having an issue.

1) reformat and install debian lite. This will reduce the changes that something from the GUI is trying to use the serial port. The GUI plus the fact that your ttyS0 and ttyAMA appear to be swapped is strange. View the ports using "ls -l /dev/ser*

2) once you have done this then run serialconfig and serialtest (with your cable). Post your results here along with the command line you are using for both programs.

3) Test with genmon and let me know the results.

Jason

nekitti, 2020-12-08:
Okay. So I gave up on this for awhile.  About 2 months ago I ordered a new RS232 To TTL Converter Module Serial Module DB9 Connector 3.3V-5.5V Arduino n HF.  I also ordered a new D-Sub 9pin Solderless Connectors DB9 RS232 Serial to Terminal Adapt QPHF.  They were on the slow boat from China and finally got here last week.  Changed them out today and it fixed it!  So even though the old RS to TTL passed the serialconfig and serialtest, it must have been bad.  Hope this helps someone else out.  Thanks for all the help offered

jgyates, 2020-12-08:
I am glad it is working now. Thanks for letting everyone know how this was resolved.

nekitti, 2020-12-08:
Thank you!

Thanks,
Rob