jgyates/genmon #390: Does not start after path change & "genmonmaint.sh -i"

skipfire, 2020-05-28:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

genmonmaint.sh -i updates crontab path entry if necessary

### Actual Behavior

genmonmaint.sh -i did not update path in crontab entry

### Steps to Reproduce (including precondition)

Install, alter the username, reboot (does not start as I expected), install again and it says 
"Crontab already contains genmon start script:" and then displays the crontab entry, which has the original username.  Manually updating /var/spool/cron/crontabs/root did fix the startup issue.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: v1.14.10

jgyates, 2020-05-28:
I am not sure what you mean by "alter the username", can you give more details?

The crontab is modified for the root user, however the python scripts are typically in the /home/pi/genmon directory

skipfire, 2020-05-28:
I renamed the pi account for security, so I renamed pi to adminpi and correspondingly renamed the home directory from /home/pi to /home/adminpi.  That of course changed the genmon location invalidating the crontab entry which still pointed to /home/pi/genmon/startgenmon.sh.  Running genmonmaint.sh for -u or -i did not replace the invalid crontab path with the valid one, it just makes sure that crontab contains startgenmon.sh.  Based on the code I think you could even just manually comment the line in the crontab and if you run genmonmaint.sh again it would tell you it is in the crontab already.

jgyates, 2020-05-29:
I will add this to the list, but in interim I will add this to the known issues list in the wiki. I certainly did not code the script to be able to handle the moving of the files to a different directory. I doubt I will be adding any new features until I have finished the encryption research that I am doing now so for now I will just document this. 
I am going to close this issue for now but feel free to post here if you have other issues on this topic and thanks for pointing out this behavior.

skipfire, 2020-05-29:
If I figure out enough bash or python to do it I will submit a PR, but I'm more of a Java/C/C# guy.

closed 2020-05-29 by jgyates