jgyates/genmon #392: [Feature request] LDAP Support

skipfire, 2020-05-29:
Is there interest in LDAP support?  I have a change working for validating username & password against LDAP and then treating a valid user as an admin which covers my primary desire.

I'd like feedback on LDAP integration before I submit a PR.
Use groups for admin & read only?
Use specific user names for each?
Require and/or allow single-user auth and LDAP to both be active?

jgyates, 2020-05-30:
I can't speak for everyone, but I would imagine that this would be more useful for companies rather than individual homeowners. If anyone wants this please chime in.

If you do decide to issue as PR, make sure it is disabled by default so this will not interfere with anyone's existing settings.

skipfire, 2020-05-30:
@jgyates right now it only enables if you enter in an ldap server value, no value has no impact.

jgyates, 2020-06-03:
LDAP support added in V1.14.12 (not enabled in web interface, must be enabled via genmon.conf)

closed 2020-06-03 by jgyates