jgyates/genmon #398: general question about updates

grsthegreat, 2020-06-03:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

Just wondering, i have received notices that there have been 2 updates to the genmon software since my last update. where can i look to see what is being updated with the current updates... i hate updating something thats working so good.....

i have V1.14.10 running now

thanks

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-06-03:
On the About page of the web interface is the button to update to the latest version. Just below the update button is a link that will display the change log.

grsthegreat, 2020-06-03:
Got it...thank you

closed 2020-06-03 by jgyates