jgyates/genmon #399: Remote Start Register and Value?

granttoth, 2020-06-03:
First off... Really fantastic software!

I don't have a specific issue and I didn't see a way to direct message you. I apologise if this is not the appropriate way to contact you.  Please allow me to ask you a question about control with the H-100.

Unfortunately I'm in a position where a Raspberry Pi won't be ideal for my situation but a simple microcontroller will be great.  I've programmed an Arduino using the registers listed and so far everything is working well except for the remote start, remote stop and acknowledge alarms. Can you clarify this for me?  

I see you use the write multiple registers command, there is an input listed at 0x0080 and the bit in that register at 0x1000. This does not work for me.  I noticed another section where you have what looks to be start, stop, and start transfer code that builds the data to send based on the value, value2, and value3 data.  Then your writing 3 registers.  

I've tried all sorts of various combinations of writing to these registers in an attempt to start, stop and acknowledge alarms, but I'm missing something here.  Do you have any advice?  Oh I should mention that the generator does work as expected when using a Raspberry Pi with Genmon.  Also my Arduino is writing as expected when I test it with a modbus simulator... I'm simple missing something.

Thanks for your time!

jgyates, 2020-06-03:
the 0x0080, 0x1000 are just status bits. You want to look at the code in generac_HPanel.py line 2056. 

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_HPanel.py#L2056

This code is writing three consecutive registers in one modbus transaction. The starting modbus address is 019c (START_BITS). You will notice that the code has more functionality that is exposed from the UI as not all of this has been validated and some of it depends on the type of transfer switch that is used by the H100 or Gpanel generator. You can look at the data like this:

Reg - > DataHigh: DataLow
019c  > ValueHigh: ValueLow
019d - > Value2High: Value2Low
019e -> Value3High: Value3Low

The modbus packet would look like this:

ADDRESS,COMMAND, REGISTER, NUM_REGISTERS,  NUM_BYTES, DATA, CRC

Using the values from the code (Value, Value2, Value3) the packet should look like this:
0x64 - Modbus Address
0x10 - Write Modbus register command
0x01 - High byte of starting register 019c
0x9c - Low byte of starting register 019c
0x00 - Number of registers high byte (0x0003)
0x03 - Number of registers Low byte (0x0003)
0x06 - Number of Low (0x06)
Data - Value High byte
Data - Value Low Byte
Data - Value2 High byte
Data - Value2 Low Byte
Data - Value3 High Byte
Data - Value3 Low byte
CRC
CRC

you can contact me at generatormonitor.software (at) gmail

liltux, 2020-06-03:
@granttoth would you be interested in sharing your Arduino code?

granttoth, 2020-06-05:
Yes absolutely.  I had planned on reporting back here once I add the control commands, and I can add my Arduino sketch too.

Heads up, I'm not great at writing code, however it's working great so far. I would love some more eyes on it to help fine tune things.  I'm going to try to find the time to add the start, stop, and alarm reset commands tomorrow.

granttoth, 2020-06-12:
@liltux I finally got around to this.  Here is what I have so far.  I started adding a struct array with all the registers and such, however I'm debating on if I need to do that or not.   If you have any advice or thoughts, please jump on in and make some changes.   

`/*
    This sketch is intended to work with a TTL to RS232 shield/board and a relay board or two.
    You will also need some sort of relay/button to trigger the start, stop, and to reset the alarms.

    Make sure to install the ModbusMaster library in the Arduino IDE.
    Tools > Manage Libraries

    The registers in the testing function are used to force the generator to show results. Pulling a sensor and such.  I haven't tested this part yet.

*/

#include <ModbusMaster.h>

int startPin = 19;
int stopPin = 18;
int ackAlarmPin = 17;

// Timer used to set the metering interval
const unsigned long eventInterval = 3000;
unsigned long previousTime;

// Generator slave ID
const int genId = 100;

// Initiate ModbusMaster object
ModbusMaster node;

void setup()
{
  
  // Pins 0 & 1 are used for the serial communication
  // Set pins 3 through 16 as outputs
  for (int thisPin = 2; thisPin < 16; thisPin++) {
    pinMode(thisPin, OUTPUT);
    // Make sure pins default off
    digitalWrite(thisPin, LOW);
  }

  pinMode(startPin, INPUT_PULLUP);
  pinMode(stopPin, INPUT_PULLUP);
  pinMode(ackAlarmPin, INPUT_PULLUP);

  Serial.begin(9600);
  node.begin(genId, Serial);
}

void loop() {
  unsigned long currentTime = millis();

  inputPoll ();

  if (currentTime - previousTime > eventInterval) {
    registerRead ();
    //testing ();
    previousTime = currentTime;
  }
}

void inputPoll () {

  // Input Pins
  int startPinValue = digitalRead(startPin);
  int stopPinValue = digitalRead(stopPin);
  int ackAlarmPinValue = digitalRead(ackAlarmPin);

  if (startPinValue == LOW) {
    startGen ();
  }
[... 211 lines cut ...]
    if ((data[2] & 0x4000) == 0x4000) {
      digitalWrite(7, HIGH);
    } else {
      digitalWrite(7, LOW);
    }

    // Throttle Position Fault
    if ((data[3] & 0x8000) == 0x8000) {
      digitalWrite(8, HIGH);
    } else {
      digitalWrite(8, LOW);
    }

    // Emergency Stop
    if ((data[6] & 0x0200) == 0x0200) {
      digitalWrite(9, HIGH);
    } else {
      digitalWrite(9, LOW);
    }

    node.clearResponseBuffer();
  }
}

`

liltux, 2020-06-14:
@granttoth thanks!  I will take a look and see what I can do.

liltux, 2020-06-15:
@granttoth is it ok that I can email about the arduino stuff?  If so you can reach me at liltux [at] liltux dot com.

granttoth, 2020-06-15:
Yeah no problem. Heads up though I'm not great at this stuff, but I'll help however I can. granttoth [at] gmail.com is me.

closed 2020-06-22 by jgyates