jgyates/genmon #400: question about estimated fuel guage

grsthegreat, 2020-06-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

now that i have added a wifi propane tank monitor to my system, is there any way to get rid of the estimated fuel use icon on the dashboard. if not, no problem, i was just wondering .

the tank utility monitor shows me actual fuel remaining, so i dont need the estimated value.

thank you in advance

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-06-04:
on the settings page set the tank size to zero.

grsthegreat, 2020-06-04:
that did not work. that actually made both estimated and external tank show azero, and issued an alwert on my phone that i lost communication. i restored it to 1000 gallons and both meters are communicating again

jgyates, 2020-06-04:
once you set the tank to zero and hit save it will restart genmon. Once it is restarted you will also need to refresh your browser. The web app does not expect the tank size to change dynamically  so you have to restart the web app also.

grsthegreat, 2020-06-04:
ok, thank you. that did it. ill close this thread.

closed 2020-06-04 by grsthegreat