jgyates/genmon #401: Power output does not reflect actual output

skipfire, 2020-06-06:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Power output changes as power outputs

### Actual Behavior

Power output meter stays between 21.5 and 22.5 any time it is running, with or without load, with or without the air conditioners, etc.

### Steps to Reproduce (including precondition)

Using the generator

[image]

[image]

It is a 22kW generator, but on NG so realistically it should be maxing around 19.5kW under full load.
One note from the genmon.log was that it did not find the model online.
2020-06-06 09:36:13,199 : Looking up model info on internet
2020-06-06 09:36:14,844 : Found: Model: G0070432
2020-06-06 09:36:14,845 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:757

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: 1.14.12

liltux, 2020-06-06:
@skipfire the 22 kw air cooled units do not have CT, current transformers for measuring current.  I am not sure what is being held in that register as it may be a calculated value based on other information.  Knowing this, on 22kw air cooled, you cannot accurately read amperage measurement or KW from the controller.

skipfire, 2020-06-06:
Ok, bummer.  Thanks for the info though.

liltux, 2020-06-06:
20kw and below air cooled units do have CT, if the correct register is mapped and the CT are calibrated then I would suspect possible accurate current measurement on those units.

closed 2020-06-06 by skipfire