jgyates/genmon #402: Installation Issues

pridesleap, 2020-06-16:
When installing GenMon via the genmonmaint.sh, I noticed several erorrs relating to the PIP installation of the various python modules. I didn't copy the output but I beleive what happened was the script attempted to first install python-pip but couldnt find the package. Therefore the subsequent installation of the various pythn packages via PIP failed. I attempted to manualy install PIP, and recieved the following error:

**E: Unable to locate package python-pip**

python-pip doesn't appear to be in the standard sources for ubuntu. Which raises two questions:

1. What source should I use in order to get it installed?
2. Do I re-run genmonmaint.sh in order to make sure all the python packages are installaed correctly?

jgyates, 2020-06-16:
I am guessing that ubuntu has deprecated python2 and is using python3 as the default. To confirm this type:

      python -V

This should return the python version. If it is 3.x then you have python3 as the default. The Pi has python2 as the default.

If you ran the install script and selected install genmon on boot, then you need to either remove the crontab entry or edit it.

if you type :

    sudo crontab -e 

(you may need to select nano as the editor if prompted)

Then remove this line:

    @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

or modify it to this line:

     @reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh -p 3 start

to save the file type:

Ctrl+X then Y, then Enter

Then you can run the script again using python3

    bash ./genmonmaint.sh -i -p 3

pridesleap, 2020-06-16:
Yes, that was it.  I executed the Python2 version of the install script. I did as you recommended and the installation went better. My manual insallation of PIP failed as I was trying to install the P2 version and as you said, its been depreciated by Ubuntu.

However, I am still not able to start genmon, either automaticaly through the cron or manualy. When I attempt a manual start, I get the following output:

_user@genmonserver:~/genmon$ ./startgenmon.sh start -p 3
Using Python 3.x...
Starting genmon python scripts
Warning: required library ldap3 not installed. Attempting to install....
Error installing module: ldap3: [Errno 2] No such file or directory: 'pip'
Error: unable to install library ldap3
Error check system readiness. Exiting_

So I attempted to manualy install ldap3 via pip3 and got this:

_user@genmonserver:~/genmon$ sudo pip3 install ldap3
Requirement already satisfied: ldap3 in /usr/local/lib/python3.8/dist-packages (2.7)
Requirement already satisfied: pyasn1>=0.1.8 in /usr/lib/python3/dist-packages (from ldap3) (0.4.2)_

Any suggestions?

jgyates, 2020-06-16:
There are two thing going on here:

1) for some reason the program genloader.py is not passing the test of the ldap3 test. This moduble should have been loaded by genmonmaint.sh during the install.
2) When genloader.py detects this condition it fails when it attempts to load because of a bug in genloader.py.

I check in V1.14..13 which should fix the bug in genloader. I am not 100% sure that this will fix the problem as I am not sure what is causing number 1 listed above but let's try the update. From the genmon directory type:

      bash ./genmonmaint.sh -u

This will update to the latest version. From there you can just type:

    ./startgenmon.sh start -p 3

Let me know how it goes.

pridesleap, 2020-06-16:
**I ran this  (should it be bash ./genmonmaint.sh -u -p 3)?:**

_u_ser@genmonserver:~/genmon$ bash ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.   Continue? (y/n)  y
Updating genmon...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Total 7 (delta 6), reused 7 (delta 6), pack-reused 0
Unpacking objects: 100% (7/7), 1.25 KiB | 106.00 KiB/s, done.
From https://github.com/jgyates/genmon
   20a0581..f0072ce  master     -> origin/master
HEAD is now at f0072ce V1.14.13
[sudo] password for user:
Done.__

**Then I started genmon and recieved the following output:**

_user@genmonserver:~/genmon$ ./startgenmon.sh start -p 3
Using Python 3.x...
Starting genmon python scripts
Warning: required library ldap3 not installed. Attempting to install....
Starting....
Starting /home/wille01j/genmon/genmon.py
Starting /home/wille01j/genmon/genserv.py_

**I the rebooted the server
I then listed processes to check that that cron started genmon:**

_user@genmonserver:~/genmon$ ps -ef |grep genmon.sh
user    1720    1293  0 16:41 pts/0    00:00:00 grep --color=auto genmon.sh_

**I then tried to access web page:**

_IP_APPRESS:8000_

.... and got nothing.

Interestingly, I used git command to download the latest files just last night. I was wondering why the update was necessary.

jgyates, 2020-06-16:
This should be 

     ps -ef |grep genmon.py

not 

     ps -ef |grep genmon.sh

Would be helpful to know these three files output:

/var/log/genmon.log
/var/log/genserv.log
/var/log/genloader.log

jgyates, 2020-06-16:
What version of ubuntu are you running?

pridesleap, 2020-06-16:
just updated my previous comment. I caught the error on the process list here is the revised output:

:~/genmon$ ps -ef |grep genmon.py
user    1852    1293  0 17:08 pts/0    00:00:00 grep --color=auto genmon.py

I'm running:
Description:    Ubuntu 20.04 LTS
Release:        20.04
Codename:       focal

How would you like the log files?

jgyates, 2020-06-16:
you can attach them to this thread if you are using the web interface from a desktop browser. Or you can email them to generatormonitor.software(at) gmail

pridesleap, 2020-06-16:
At this point I should note that other than installing genmon, I haven't configured anything yet, nor do I have a functioning serial connection.

Here are the log files:
[genserv.log](https://github.com/jgyates/genmon/files/4788117/genserv.log)
[genmon.log](https://github.com/jgyates/genmon/files/4788118/genmon.log)
[genloader.log](https://github.com/jgyates/genmon/files/4788119/genloader.log)

jgyates, 2020-06-16:
Based on your logs there are two issues, one of them is the reason that it is not loading.

in genmon.log there is this message:

2020-06-16 17:05:03,961 : Error opening modbus device: Error opening serial device. : generac_evolution.py:235
2020-06-16 17:05:03,962 : Possible cause is invalid serial port specified.
2020-06-16 17:05:03,962 : Error getting thread name in KillThread: ComWatchDog
2020-06-16 17:05:04,961 : Generator Monitor Shutdown

This means the serial port specified (the default is /dev/serial0) is not present. You can change this by editing this line:

      port = /dev/serial0

in /etc/genmon/genmon.conf

The second problem is in genaloder.log

Warning: required library ldap3 not installed. Attempting to install....

For some reason genloader is not detecting ldap3 is installed and it attempts to install it. The new update is not causing an error now so this is more of an anomaly than a hard error since genmon continues to load.

If you continue with your setup of the serial port or serial over IP you should be fine.

jgyates, 2020-06-16:
also, unless you are planning on using LDAP for authentication to your web interface for genmon the ldap3 module will never be loaded so you can ignore that error for now.

pridesleap, 2020-06-16:
Ok, I'll work the serial port issue and push forward. I have the serial to IP converter now so I can get that configrued and try to get this issue resolved.

Im not really familiar with LDAP anyway so I can probably ignore it for now.

Should I close this issue then?

jgyates, 2020-06-16:
we can leave it open for a few days to see if you have any other ubuntu specific issues.

pridesleap, 2020-06-16:
will do, thanks for the help!

jgyates, 2020-06-16:
FYI, I just loaded a virtual machine with Ubuntu 20LTS and serial over IP and everything worked great. I don't even get the ldap problem.
[image]

pridesleap, 2020-06-16:
I'm running Ubuntu Server 20.04, not Desktop. Installed Xubuntu-core for a desktop. Maybe the difference is somewhere in between the server install and xfce. I'll try setting up an Ubunu Desktop install if thats what it takes.

Do you have a reference to setting up serial over IP anywhere in the wiki?

jgyates, 2020-06-16:
https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

pridesleap, 2020-06-16:
ah, "see 6 more pages". Still learning........

pridesleap, 2020-06-19:
Out of curiosity, can installing this over SSH cause any problems? I exclusively log into my server remotely and was wondering if this is something that should be done localy instead.

liltux, 2020-06-19:
@pridesleap I only ever operate my Pi's headless via ssh, never had a problem.

jgyates, 2020-06-19:
same here. I rarely use a keyboard or monitor with my pi. 

Installing with ssh enabled is easy once you know the trick.

https://hackernoon.com/raspberry-pi-headless-install-462ccabd75d0

pridesleap, 2020-06-19:
Ok, well I got it working. I think it was all user error. It occured to me after I asked that most PI installations will probably be over SSH anyway but it was worth an ask. Heres what I did:

- I re-built my server from scratch. This time I didn't install any desktop. It's a pure Ubuntu Server 20.04 install. Updated and upgraded apt as needed.

- I re-installed genmon following your directions. This time minding the references to Python3. I also updated genmonmaint.sh BEFORE installation. The installation took longer than before and there weren't any noticeable errors. ALthough I do still get the error regarding LDAP.

- I configured my serial TCP settings targeting the RS232 to TCP converter. I'm not getting any live data yet but thats because I'm not connected to my generator.

- The server is running though and I was able to load the web page from another client on my network.

Next step is to connect the generator and see if I get valid data. However for now, I think my issue is resolved. Thanks for the help with this. This is going to be very helpfull.

pridesleap, 2020-06-19:
Ok, well a new problem. I lost the gauges on the Status page after a server re-boot.

[image]

jgyates, 2020-06-19:
try refreshing your browsers. If you open the web interface before the software has had a chance to query the generator this can occur, you just need to refresh your browser. also, unless you are communicating with the generator those gauges are place holders anyway.

pridesleap, 2020-06-19:
I tried emptying my cache, restarting the browser, restarting the client and tried different clients. The gauges disappeared from all of them. They were there the first time I logged in though but disappeared after the reboot. Anyway I also noticed that genmon automaticaly starts on boot, even though I removed the command from the cron. I did have the installer add it to the cron but then I removed it after. Still auto starts.

I'm going to pause a bit and pick it up again this weekend.

jgyates, 2020-06-19:
did you remove it from crontab for the root user?

    sudo crontab -e

you may need to select nano if prompted, then remove the line, save the file (Ctrl+X, then y, then Enter)

Unless you are communicating with the generator I would not focus on the gauges.

pridesleap, 2020-06-22:
Agreed, Im not going to worry about functionality of the app until I am actualy connected to the generator. That being said, there is a clear difference between genmon instlling and running on Ubuntu Server 20.04 vs. Ubuntu Desktop 20.04. I rebuilt my server using the desktop version (althoguh it was a minimal installation). I saw the same things you did. Installation went smoothly, no errors. There even doesnt seem to be any issue with LDAP at this point. I can access the genmon web pages from the Localhost or from a remote client either on the local network or through my VPN. I connected a MODBUS simulator to the Serial to TCP gateway and I am getting valid data through to my linux server so I should be ok with the current configuration

I really wish I could figure out what about Ubuntu Server causes my issues as I like lightweight servers without desktops. Ubuntu can be a bit heavy. Either way I'm going to run with this and just install the server packages I need. I consider my specific issue closed as I believe it was an issue with the version of OS I was running and not specificaly the app itself.

closed 2020-06-22 by jgyates