jgyates/genmon #403: Genmon Use of 003a instead of 05f4 and 05f5 for Load Current Values

sefs85, 2020-06-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon should read the CT1 and CT2 registers to determine load current. 05f4 and 05f5

### Actual Behavior

Gemon is reading 003a, which I believe is actually the position of the stepper motor controlling the throttle. This is why it has a significant value when you are running no-load, but also appears to be closely related when loads are put on or taken off of the generator. 

The best solution would be if we could have both L1 and L2 voltages (~120V), however, since the generator only reads L1-L2 voltage (~240V), the closest calculation that should be done is Approximate_240V_Load = (CT1+CT2)/2. This would then properly calculate into the kW Power.

### Steps to Reproduce (including precondition)

Anytime the generator is running. 

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 6551 Generac 22kW (I have retrofitted current transformers into my 22kW, as it doesn't come with them stock. I can confirm when I go into the dealer menu it properly reads the current values.)
- Generator Registers: 05f4 and 05f5
- Genmon Version: 1.14.13

jgyates, 2020-06-17:
Hi @sefs85 

On the registers page, 5F4 and 5F6 are labeled CT for Evo air cooled, but my confidence that this is the actual value is not that high. I tried using 5f4 and 5f5 before, but the result was not what I expected. I will caveat this with the fact that I was not doing this test myself so I am willing to entertain changing this, I would just need more data and a description of how we validate the assumption that 5f4 and 5f5 are actually the CT outputs.

sefs85, 2020-06-17:
I can do some more definite investigation with my function generator and the CT’s seeing what the math equates to in those registers. I believe it corresponds directly through *10 for the value, but I’ll do some testing to find out for sure.

sefs85, 2020-06-18:
I did some testing, and confirmed that the value that is read when you are in the dealer menu on CT1 or 2, which reads the amps out directly (already compensated for the calibration factor), are the values that are in 05f4 and 05f5. The only conversion needed is /10. 

I did some trial modifications to the code in "generac_evolution.py" starting at line 3000 and got accurate results for power output.

            elif self.EvolutionController and not self.LiquidCooled:
                Value = self.GetRegisterValueFromList("05f4")
                DebugInfo += Value
                if len(Value):
                    CurrentFloat = int(Value,16)
                    Value = self.GetRegisterValueFromList("05f5")
                    CurrentFloat = CurrentFloat + int(Value,16)

Then I set the current divider to 20 and current offset to 0 in the advanced menu in genmon. The divide by 20 gets me a /10 to compensate for the register being in *10, and then /2 for the average of the two currents. 

I still need a way to do some more definite testing on my suspicion of 003a being throttle percentage, but that is my best guess at this point.

jgyates, 2020-06-18:
@sefs85 

this is great info! I will add this to the code. I will likely put an option in to revert to the old way as it is possible that the CTs may need to calibrated differently on different models.

Out of curiosity, how did you determine the CT specs when you retrofitted your 22kw?

sefs85, 2020-06-18:
On page 80 of the service manual, it has a table which states the approximate voltages across the input pins on the controller. I took my DMM and measured the impedance, and got about ~18 omhs. It stated that 100 amps would equal 1.510 VAC measured. The closest commercially available ratio was 1000:1. So I gave it a shot and it worked perfectly. I didn't even need to adjust the factor set calibration factors on the controller of 1483. The pins I got from digikey weren't exactly correct, but they did work. I got these split core CT's from Amazon, one for each line. 

[https://www.amazon.com/gp/product/B01LWN37KS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1]

jgyates, 2020-06-18:
I checked in version 1.14.14 which includes this change you suggested above. 

@everyone, please see changelog.md on the about page or here:

https://github.com/jgyates/genmon/blob/master/changelog.md

Thanks for the addition @sefs85, let me know if this works for you. I made the default divisor 20 if using this new (default) way of doing this.

sefs85, 2020-06-19:
@jgyates I tried it out and it appears like it's all working. Thanks for the update!

grsthegreat, 2020-06-19:
I tried it on my 22 kw evol 1 unit and now it shows no value in power output during actual transfer testing. It did not work properly in old mode as it showed 8 kw load during non load testing

sefs85, 2020-06-19:
@grsthegreat You have to add CT's to a 22 kW unit, it did not come with them from the factory.

grsthegreat, 2020-06-19:
Where would the ct’s be hooked up to...
 How does the controller calculate current draw without ct’s.

sefs85, 2020-06-19:
Pins 9 and 10 on the J1 connector for the A CT, and pins 6 and 14 on the J2 connector for the B CT (Refer to page 296 of the service manual). 

I don't believe the old value was calculating current draw, but was actually a value tied to the throttle; but that has yet to be confirmed.

grsthegreat, 2020-06-19:
ok, cool. i think ill order the set of ct's that you used. 

in servicing older geneators ive seen the ct's on the units. i never realized that they were not installed in the 22kw units. 

did you have to do any changes to any settings after installing the ct's?

grsthegreat, 2020-06-19:
also, my service manual ends on page 290 for most current one i have 0H1972 rev M and 10000041488 revB

sefs85, 2020-06-19:
You'll need to get some pins to insert into the connectors as well (as they are currently empty). I ordered these (https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/770904-1/A25684CT-ND/409446), but they were not the same as the stock ones and didn't fit perfectly. I'd try to find ones that fit better. There were no settings that needed changed, everything just worked when I hooked it up. You can read out the current values directly if you go into the dealer menu and press enter when you're on the screen where you set the CT calibration numbers.

For the manual, mine is 0H9172 rev J, I'll have to get a more up to date manual. But the section was called "Appendix A, Table A-1 Evolution Connector Pin Descriptions and Pinout".

skipfire, 2020-06-19:
@jgyates in the readme you say for anyone who doesn't have the 22kW it is recommended to use the new method, since the 22kW doesn't have the CTs by default does it matter?  It doesn't show anything accurate now for power use.

grsthegreat, 2020-06-19:
Generac lists that part as 0J1702 molex connector. at 0.11 each, i guess ill get a few

sefs85, 2020-06-19:
@grsthegreat Post back how they work out. I'd like to swap out my pins for the right ones.

jgyates, 2020-06-19:
You are correct, the 22kW unit, as it ships will continue to not report the current output.

The point I was trying to get across is that if you have a 22kw, this software update will not correct the fact that you do not have CTs installed. 
22kw

grsthegreat, 2020-06-19:
sefs85... i was just placing an order for a replacement controller for a customer and some stock items. i added 12 of the molex connectors to the order. if they work out ill mail you 4 for free.

grsthegreat, 2020-06-19:
> 
> 
> You are correct, the 22kW unit, as it ships will continue to not report the current output.
> 
> The point I was trying to get across is that if you have a 22kw, this software update will not correct the fact that you do not have CTs installed.
> 22kw
jgyates.....i see that now. i never knew that the 22jgyates.....i see that now. i never knew thakw units didnt have ct's.

i ordered 2 cts and some molex connectors. well get this up and running soon....mind you, i ordered parts from Generac so it will take awhile. 

they have the slowest and most expensive shipping out there.
ill get the cts from amazon in a few days.

sefs85, 2020-06-19:
> i ordered 2 cts and some molex connectors. well get this up and running soon....mind you, i ordered parts from Generac so it will take awhile.
>
> they have the slowest and most expensive shipping out [...]

Make sure you have some extra wire to extend the leads from the CT's. I added a few feet of 18AWG with some butt splice connectors.

grsthegreat, 2020-06-19:
> Make sure you have some extra wire to extend the leads from the CT's. I added a few feet of 18AWG with some butt splice connectors.
thanks for the heads up. ill keep you informed when its up and running,... and ill dend some molex connectors if they fit good.

greg

skipfire, 2020-06-19:
It would be great if someone with that manual could post the critical details for connecting the CTs to the controller.  I'm not asking for a page scan or anything that would have copyright problems, just pinouts and such. Maybe even a wiki page?  I've searched online without luck at finding an instruction manual that even gets to 200 pages without luck, I'm guessing it's a dealer/tech manual?  I've also been searching for adding the CTs after the fact, and have found nothing.  Any details to clarify at least the low voltage end would be appreciated.  I'm good with the other end.

liltux, 2020-06-19:
What controller?  I could get you pin locations.  I do not know though the unit of measure.  I e 0-5 amp or voltage.

skipfire, 2020-06-19:
Evolution 2

jgyates, 2020-06-19:
FYI, you can purchase PDF diagnostic manuals for Evo1 and 2 here

https://www.zillerelectric.com/collections/generac-manuals

sefs85, 2020-06-19:
Page 300 of this manual has the pin outs. 

https://www.generator-parts.com/manuals/home-standby/0H9172.pdf

skipfire, 2020-06-19:
> Page 300 of this manual has the pin outs.
> 
> https://www.generator-parts.com/manuals/home-standby/0H9172.pdf

The secret books!  Thanks.  :-)

sefs85, 2020-06-19:
> > Page 300 of this manual has the pin outs.
> > https://www.generator-parts.com/manuals/home-standby/0H9172.pdf
> 
> The secret books! Thanks. :-)

At the low low price of free 99!

skipfire, 2020-06-19:
> > > Page 300 of this manual has the pin outs.
> > > https://www.generator-parts.com/manuals/home-standby/0H9172.pdf
> > 
> > 
> > The secret books! Thanks. :-)
> 
> At the low low price of free 99!

Rockin!  Thanks!  Great resources I didn't know about, thanks for both links all!

skipfire, 2020-06-19:
> FYI, you can purchase PDF diagnostic manuals for Evo1 and 2 here
> 
> https://www.zillerelectric.com/collections/generac-manuals

Thanks, I figured these kinds of books were out there, I just hadn't found them.

grsthegreat, 2020-06-20:
[image]

grsthegreat, 2020-06-20:
[image]

grsthegreat, 2020-06-20:
[image]

grsthegreat, 2020-06-20:
you can see how the units ct1 & ct2 connect. the wire numbers are printed on the molex connector, however they are not labeled J1,J2,etc. you need to count the wire connections.

also, ended up ordering 24 of the connectors. if anyone else needs any i can pop them into an envelope....if they work like they should.

liltux, 2020-06-20:
@grsthegreat and everyone, I just want to be clear, if you are looking at the manuals be sure you are looking at the correct SD/WD.  The Evo1 and EVO2 have different pinouts.  Just want to point out, the pic listed above is Evo1.  Just be sure you are looking at the right pinouts for your controller.

grsthegreat, 2020-06-20:
[image]

grsthegreat, 2020-06-20:
sorry...i did not catch the note that he had an evolution 2 controller. heres the schematic for the evolution 2

grsthegreat, 2020-06-20:
[image]

**THIS IS THE EVOLUTION 2 CONTROLLER ONLY**

jgyates, 2020-06-22:
I am going to close this issue however feel free to post here if you have questions or comments on this topic.

closed 2020-06-22 by jgyates

jwtaylor310, 2020-06-22:
grsthegreat: If you still have extra connector pins I would love to get 4 of them.  I'll be more than pleased to pay whatever they cost you, postage, etc.  If you would send me contact information at jtaylor08@tecosales.com I'll replay with my shipping address, etc.  Thanks!

grsthegreat, 2020-07-07:
i still have not rec eived the pins from generac. they were backordered but i expect shipment within a few days. if you want 4 email me   grsaec1@gmail.com

liltux, 2020-07-07:
These are minifit jr connectors by molex and use an 18ga pin male/female.  Can be purchased through other venders such as:
https://www.digikey.com/product-detail/en/molex/0444851211/WM8749CT-ND/3175868
FYI

grsthegreat, 2020-07-07:
I ordered 24 of the connectors from generac...
Since i was ordering other stuff anyways. But they were like .12 apiece.

grsthegreat, 2020-08-04:
I finally got the molex connectors. If anyone needs some shoot me an address or email.

grsthegreat, 2020-08-05:
I was finally able to hook up the ct's to the controller as Generac took 6 weeks to ship the molex connectors. I ran a test run using genmon monitoring and a clamp on meter for comparison. the Genmon gave an exact reading of the average load between both legs as calculated using my clamp on meter. i am very happy camper now.

I like being able to see exact load on generator. I am also VERY pleased with the Hyper Sure Start that i just recently placed on my 4 ton AC compressor. I used to see a micro 97 amp load when the compressor kicked on. today i saw 22 amp. The generator didnt even burp.

sefs85, 2020-08-05:
I'm glad you were able to get it working. I really like the accuracy as well!

jwtaylor310, 2020-08-11:
I just received the molex pins that I ordered June 24th.  I used them to terminate the wires from a pair of Current Transformers that I previously purchased for installation in my generator.  When I opened up the generator to install them I was surprised to find existing wiring attached to the pins on the J1 and J2 plugs that are used for CT's.  I was able to trace the wires and discovered that there are already two (blue) factory-installed CT's in my generator.  This is a 12KW unit with Evo2 purchased in April, 2019.  At the time of installation, Genmon reported power output of 21.91 kW at all load levels and I assumed that was due to the generator not having any CT's installed.   Today, without making any wiring changes I turned off the "Disable Power/Current Display" switch in Genmon's settings and it now displays valid  power level that track the load on the generator.  Presumably, the recent register change fixed the problem.  So:

1. Evo2 owners who have disabled the power display in the belief that their generator doesn't have Current Transformers should try re-enabling the display.  It may be that your generator DOES have CT's after all.

2. If it turns out your generator truly doesn't have CT's, I have no use for the pair that I purchased for my generator.  I have already crimped on the molex pins so they are ready for installation.  Free to the first person requesting them!  I also have about a dozen extra molex pins free to anyone requesting them.

A remaining mystery is where in the power circuit the CT's are connected.  In my generator they are physically installed on wire loops that are buried behind the "Customer Connection Area" where visual inspection is difficult.  The wires going through the CT's are smaller diameter than the wires appearing on the generator side of the "E1" and "E2" main terminal block.   The schematic indicates that they CT's should be on the cables "11" and "41" running from the stator to the main power circuit breaker.  It seems odd that the wiring on the stator side of the breaker would be smaller than the wiring on the E1/E2 user side of the breaker.  Nevertheless, the power sensing appears to work correctly so this is just a matter of curiousity.

Here's a photo of one of the CT's in my generator:
  
[image]

liltux, 2020-08-11:
All Evolution have CT, except the 22 KW.  CT1 399A and 398A should be on wire 11.  CT2 399B and 398B should be on wire 44.  The wiring from the stator is of different type rating wire and insulation.

jwtaylor310, 2020-08-11:
Hi, lilux;

My generator _is_ a 22 KW Evo2.  That's why I was surprised to find that it has CT's factory installed.

Thanks for the location and wire rating info.

grsthegreat, 2020-08-11:
22kw arnt supposed to have them, only the 20 and less. Weird

grsthegreat, 2020-08-11:
Also, i doubt it matters which side of breaker you mount ct's to. My champ
on meter reads same regardlesd. The only place dealer has to mount to is
gen side during construction.  The manufacturer dies not have to follow NEC
wire size. Heck, neither do utilities. Look at the size wires dropping off
of power poles that hook up to the ridiculous large size wires we have to
run.

grsthegreat, 2020-09-07:
Power is out......cts working great. I can now see what turning on oven snd ac unit do to my power supply.

chia9876, 2020-11-22:
> 22kw arnt supposed to have them, only the 20 and less. Weird My 22 kw doesn't have them, and the last 22kw i installed a few months back did not have them.
> 
Hello Everyone,
I'm new here and I'm in the middle of installing my 24kw air cooled gen and was planning on putting in the current transformers when I also noticed that I also have them factory installed. The 24kw just came out so maybe Generac is installing them on all gens now?!?
[image]
[image]
[image]

liltux, 2020-11-22:
Yes, I believe on 2019/2020 and newer units all have CTs.  Safe to get if your unit is 2017 and older 22KW, you do not have CTs.