jgyates/genmon #404: another issue with my genmon unit

grsthegreat, 2020-06-23:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

i lost my wireless router today and purchased a new one. i got it installed and all my computers, tank monitor and phones hooked into it just fine. But i cannot get my genmon unit to reboot. i powered it down by unhooking ground wire. and when i power it back up i get nothing. i do not see genpi login in my available wifi signals. i cannot access the unit with putty because i cant get it linked to my router. 

im at a loss

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

grsthegreat, 2020-06-23:
i left it unplugged for 10 minutes.it restarted. im closing this

jgyates, 2020-06-23:
You can make linux recheck the wifi connection at any interval, but it requires a custom script. I use a script that will periodically check the wifi connection and if it not connected it will force the wifi adapter to rescan.

This repository has the script that I use:

https://github.com/dweeber/WiFi_Check

The comments of the file describe adding this to crontab. To do this run these commands:

    sudo crontab -e

Then add

    */5 * * * * sudo /home/pi/WiFi_Check 

This assumes the file WiFi_Check is in the /home/pi directory and has the permissions

grsthegreat, 2020-06-23:
are these commands  installed thru putty? my genmon keeps having touble communicating

liltux, 2020-06-23:
Yes, has your signal changed?  You are using a new router maybe the signal is not as before.

grsthegreat, 2020-06-23:
ok. i can access my genmon with putty, and i typed

 cd genmon
 ./startgenmon.sh restart
  python ClientInterface.py

this reset unit. i can still access it thru putty but i cant access the unit at all with my computer. just says server has stopped responding

this is getting very frustrating

grsthegreat, 2020-06-23:
i have a good signal at generator. i can access interned with my ipad there. generator also has an external antenna. shows 3 bars

liltux, 2020-06-23:
So you can putty in but not pull up on the webserver?

grsthegreat, 2020-06-23:
That is correct. Snd weird thing is about 10 min ago i was able to view it on webserver. I noticed that tankutility was deactivated which is good as that add on started this entire thing. Ive decided nit to use the add on.

But when i just tried to access the generator a minute ago...cannot open page.

jgyates, 2020-06-23:
Can you describe what happens when you said "this reset unit"? Did you ssh Putty session stop responding? This is not necessarily a reset of the pi, but a problematic wifi signal. When you use Putty, every keystroke and character printed to the screen are sent over your network connection so if that is not working properly then that would cause issues.

Also, having a good signal measured form an iPad is not the same as the signal at the pi. This will let you know the wifi signal if you can ssh into the pi:

     cd genmon
     ./startgenmon.sh restart 
      python ClientInterface.py

From within the ClientInterface.py program type:

      generator: monitor 

This will display the same info as the monitor page on the web interface, which has the wifi signal strength. This page has good info on what acceptable signal strength should be:

https://eyenetworks.no/en/wifi-signal-strength/

Your router may also have info on connected clients and the quality of the connection.

grsthegreat, 2020-06-23:
Ok.  Whats happening now. I can access it by using firefox on my ipad, but not when i use chrome.

liltux, 2020-06-23:
This is sounding slightly familiar, is it possible your genmon.conf is corrupt?  Can you copy over a fresh genmon.conf to the etc/genmon/ folder.

jgyates, 2020-06-23:
what is the wifi signal strength on the Monitor page?

grsthegreat, 2020-06-23:
WLAN Signal Level : -54 dBm
        WLAN Signal Quality : 56/70
        WLAN Signal Noise : -256 dBm

grsthegreat, 2020-06-23:
Shows 4 bars

grsthegreat, 2020-06-23:
> This is sounding slightly familiar, is it possible your genmon.conf is corrupt? Can you copy over a fresh genmon.conf to the etc/genmon/ folder.

Im not exactly sure how to do that

jgyates, 2020-06-23:
that is a pretty good signal. Do you have email setup on this system (are you getting emails when genmon starts?)

jgyates, 2020-06-23:
To recopy the genmon.conf file, log in via putty:

    cd genmon
    sudo cp ./conf/genmon.con /etc/genmon
    ./startgenmon.sh restart

This will rest you base configuration (no email or any add ons). If your serial port is not the default then you will need to reset that.

grsthegreat, 2020-06-23:
i do have email, and i keep getting updates while messing with this issue. i just cleared all safari cache and webdata. after clearing it 2 times, i can now access the generator. 
ill see if this lasts over the next day or so. if not ill reset the config file like you suggested.

thank you
im pulling the last remaining hairs out of my head over this

grsthegreat, 2020-06-25:
Well...working great again. Ill close this thread...for now
Thanks for your help all

grsthegreat, 2020-06-26:
ok....same issue today as the other day. this morning i checked and all was well. this evening i get a timeout message when trying to view monitor.

i ran config file scrip as stated a few links above as i can view unit in putty. 

still cant view my monitor. but i can access it thru Putty.

grsthegreat, 2020-06-26:
Im going to take a break from this for awhile

jgyates, 2020-06-26:
ok. If you have email setup you can send me your logs from the About page on the web interface or via ssh from the ClientInterface.py program. I can see if there is anything the software is complaining about. 

      cd genmon
      python ClientInterface.py 

 
           OK : Auto, Off - Ready
           >generator: sendlogfiles
              Log files submitted
           >exit

Since this appears to be intermittent, I would guess that this is likely related to your network or browser, but that is just a guess. This issues is the closest to what you are describing. https://github.com/jgyates/genmon/issues/271

grsthegreat, 2020-07-03:
Ok...unit is driving me nuts. My new router is probably the reason. The new unit is a netgear router with WPA/WPA2 PERSONAL encryption rating. 

I can see the genmon unit using fing, but router isnt letting it in. Any idea what i should do

jgyates, 2020-07-03:
This thread has some trouble shooting for a person having issues with their router. You can see if there are any security related features that you can whitelist (tell it to ignore the ip address of the pi) or disable that specific feature.

jgyates, 2020-07-03:
Here is the link: https://github.com/jgyates/genmon/issues/271

liltux, 2020-07-03:
Try checking the dhcpcd.conf file for correct router and DNS settings if using a static IP.

grsthegreat, 2020-07-03:
I do have to use a static ip address to use router.

jgyates, 2020-07-03:
you don't have to, but using one does take out one layer of complexity, namely DHCP, which will cause your IP address to change periodically. If you do assign a static IP it must be one that is not in the range of IP addresses that your router is using for DHCP assignments, otherwise your could likely have two devices with the same IP address, which would make connecting to either of them probematic.

grsthegreat, 2020-07-03:
My isp provider is requiring it. Using fing i can see i have no conflicts

grsthegreat, 2020-07-03:
> Try checking the dhcpcd.conf file for correct router and DNS settings if using a static IP.

Pardon my stupidity...but where do i find these

jgyates, 2020-07-03:
This page has a lot of details on networking:

https://www.raspberrypi.org/documentation/configuration/tcpip/

the file you are looking for is /etc/dhcpdc.conf

grsthegreat, 2020-07-03:
ok, thank you. ill be doing some studying

closed 2020-07-03 by grsthegreat