jgyates/genmon #405: Timeout receiving slave packet

davidrule1969, 2020-06-26:
My installation of genmon that has been running happily for years with no issue, out of the blue sent an email warning that communication with my generator was lost.

I have my Generac G0065512 connected to a serial server and genmon, running on a Pi, is configured to use a "Serial Server over TCP/IP".

I can see traffic between the serial server the Pi using tcpdump.

There are no errors in /var/log/myserialtcp.log but there are lots of errors popping up in /var/log/mymodbus.log

> 2020-06-26 13:00:42,767 : Error: timeout receiving slave packet for register 005a Buffer:0
> 2020-06-26 13:00:45,799 : Error: timeout receiving slave packet for register 005f Buffer:0
> 2020-06-26 13:00:48,828 : Error: timeout receiving slave packet for register 005e Buffer:0
> 2020-06-26 13:00:51,854 : Error: timeout receiving slave packet for register 005d Buffer:0
> 2020-06-26 13:00:54,885 : Error: timeout receiving slave packet for register 020f Buffer:0
> 2020-06-26 13:00:57,913 : Error: timeout receiving slave packet for register 020d Buffer:0
> 2020-06-26 13:00:59,681 : Discarding byte slave: 00
> 2020-06-26 13:00:59,688 : Error Receiving slave packet for register 020e
> 2020-06-26 13:00:59,703 : Discarding byte slave: 00
> 2020-06-26 13:00:59,711 : Error Receiving slave packet for register 020b
> 2020-06-26 13:00:59,727 : Discarding byte slave: 00
> 2020-06-26 13:00:59,728 : Error Receiving slave packet for register 020c

### Your Environment

- Generator Model: G0065512
- Genmon Version: V1.14.14

davidrule1969, 2020-06-26:
Left it alone for about an hour and then received an email saying communication restored.  I didn't change a thing!  No new errors being reported in /var/log/mymodbus.log.

Very strange.  Will keep an eye on things and report back if problem reoccurs.

-David

jgyates, 2020-06-26:
You may want to consider the possibility of an intermittent connection. Over time, due to the vibrations from the generator, connections can become marginal unless care it taken in cable connections. I experienced an issue like this early on. I improved my cabling and the problem never occur again.

jgyates, 2020-06-29:
I am closing this issue for now. Feel free to post to this thread if you have any additional data or questions.

closed 2020-06-29 by jgyates