jgyates/genmon #406: State change emails without change?

skipfire, 2020-06-29:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

No email, or emails with identifiable change

### Actual Behavior

Emails occasionally come with no noticeable change.  Seems to be after multiple days with no other activity, but not exactly (x) days since an event.

### Steps to Reproduce (including precondition)

Just have genmon running, didn't do anything.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: 1.14.14

[Email1.txt](https://github.com/jgyates/genmon/files/4845725/Email1.txt)
[Email2.txt](https://github.com/jgyates/genmon/files/4845726/Email2.txt)

skipfire, 2020-06-29:
Submitted logs & registers.

jgyates, 2020-06-29:
This is a know issue with Evolution 2.0. See number 11 on this page:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

You can optionally have genmon ignore these values by enabling the "Ignore Unknown Values" option on the advanced settings page (double click the gear icon in the top right of the web interface).

closed 2020-06-29 by jgyates

skipfire, 2020-06-29:
@jgyates since I am getting this, is there anything I can do to help figure it out?  I have the wifi module physically disconnected and there was no access to the unit this morning.  Each time this has happened it seems to be about 10 seconds between emails.

jgyates, 2020-06-29:
one thing you can do is this:

Open the web interface and leave it open. You can monitor the registers changed while the web interface is active. Single click on the gear icon in the upper right to view the registers. You can click to different page and come back to this page and monitoring will still occur. When this happens again, you can view this page and see if any other registers change besides register 0001. Registers that change will highlighted in red. I would suggest refreshing your browser after things like your weekly exercise if this symptom has not occured yet to minimize the changes in the registers. What we are looking for is any other activity in registers besides the values in register 0001.

Let me know if that does not make sense.

skipfire, 2020-06-29:
Will changed registers stay red if they return to the original value?  I get the 2 emails in about 10 seconds, so I'm guessing it briefly alters and then goes back.

jgyates, 2020-06-29:
Yes. The background turns red and fades over the next 60 seconds for items that change recently, but the color of the letters stays red for registers that have change.