jgyates/genmon #408: how to - live register view - graph duration

nworbekim, 2020-06-30:
### Expected Behavior

a configuration that retains register data for the live register view graphs

### Actual Behavior

graph is shown but I don't understand how to observe up to 24 hours of data

### Steps to Reproduce (including precondition)

go to live register view
mouse over utility voltage, observe graph with 10 minute, 1hour, 24 hour scales

### Your Environment

- Generator Model: Nexus, Air Cooled
- Genmon Version: V1.14.15

jgyates, 2020-06-30:
The underlying data for the live view of registers is only maintained as long as the web interface is displayed in the browser. If you close the web page or refresh the page the data will be lost. To view the 24 hour graph you need to have the web page open for 24 hours.

jgyates, 2020-07-03:
I am closing this issue. If you still have questions on this topic you can post them to this thread

closed 2020-07-03 by jgyates