jgyates/genmon #409: 22kW generator showing somewhat accurate power?

skipfire, 2020-07-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I expected it to stay at 0kW (glad it isn't)

### Actual Behavior

Just posting this because I was really surprised and the new registers seem to make things work for 22kW units.  It is showing at least somewhat accurate power output, I don't have 2 clamps to get reading but it jumps by about the right amount when ACs kick on, and I was able to get it to peak and hear the load-shedding engage.  I bought the loops, but have not installed them yet.  If it makes any difference, I am on a transfer test cycle.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}
[image]

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: 1.14.15

grsthegreat, 2020-07-01:
thats not the real value. my unit did the same but when i clamped on my fluke meter, the values were way off. you need to add the CT coils. im in the process of doing this to my own 22kw unit. i have received the ct's but im waiting on the molex connectors that i ordered from generac. everything that generac mails takes forever.

jgyates, 2020-07-02:
I am going to close this issue however feel free to post here if you have additional questions on this topic.

closed 2020-07-02 by jgyates