jgyates/genmon #410: Feature request: configurable notification delay

nkm8, 2020-07-02:
I have FiOS, so I have a router and an ONT that need to be powered on for internet access to work. Unfortunately, these devices are in 2 different locations and I don't want to invest the $$$ for multiple UPSs that would only ever be used for rare 30-second outages before the generator kicks on.

It would be a great feature addition if there was a configurable delay available to set for email/SMS/push notifications. I want to be notified of outages, but the router/ONT will be offline for 30 seconds until the generator kicks on + device startup time. Delayed notifications are better than no notifications and the configuration would allow to customize for the variable start-up time for different router models.

jgyates, 2020-07-02:
I will add this to the new feature list. I am currently working on the Evolution 2 modbus issue at the moment but I will try to squeeze this in when I can.

nkm8, 2020-07-02:
I thought about this some more and wanted to propose an alternate solution. Instead of a configurable delay, add a tiered cool-down for failed requests:
- Attempt to send notification - failure
- Wait 30 seconds
- Re-attempt notification - failure
- Wait 2 minutes
- Re-attempt notification - failure
- Wait 5 minutes

I am not sure which approach is easier or which is more configurable/useful to people. I am sure there are other complexities like short outages - power cuts out, generator comes on, and power comes back on before the first notification is successfully sent. In that scenario, I'm not sure if the notifications each have their own threaded timer or if there is a central queue with one timer that controls sending all messages.

jgyates, 2020-07-02:
After taking a quick look at the code it appears that functionality similar to this is already there (apparently I forgot I already did this). If a mail message fails to send, it is queued and an attempt to resend occurs 2 minutes. This behavior repeats until successful. If you are not seeing this behavior then you can send your log files from the about page or view /var/log/mymail.log for any relevant errors.

If genmon is restarted any queued emails will be lost.

jgyates, 2020-07-06:
I am closing this issue as I believe the functionality already exist. If your testing shows differently let me know.

closed 2020-07-06 by jgyates

nkm8, 2020-07-24:
@jgyates I inadvertently tested this today when I had some electrical work done - the utility power was off for ~30 minutes. Looking at the logs, the email attempted to send 3 times, 2 minutes between each attempt. After that, the message was dropped.

I am not sure if this is intentional or not, but I wanted to clarify the actual behavior. I shared my logs, if you want to take a look. The pi0 running genmon is powered via the control board, so it never lost power during the 'outage'

jgyates, 2020-07-24:
what day did you send your logs? I have not received any since the 21st.

nkm8, 2020-07-24:
@jgyates Earlier today - although after retrying, I notice there isn't any feedback - I click 'Submit Logs' on the About page - then Ok on the modal dialog - then nothing.

jgyates, 2020-07-24:
sounds like your outbound email is not working. this is needed for log submission. You can test your email from the settings page -> outbound email , then Test Email Settings

jgyates, 2020-07-24:
Another thought is that if your pi is using a dynamic IP, if your router (DHCP server) was power cycled during the outage then your router may be assuming your the pi's IP address is used by another device on your network. If you email starts working if your pi is rebooted, this may be the problem.

nkm8, 2020-07-24:
@jgyates Test email worked fine - pi hasn't been restarted all day - static IP is assigned and reserved at the DHCP level. Here are the logs I saw earlier today before I tried the 'send logs' feature:
```
pi@pi0genmon:/var/log $ tail mymail.log
2020-07-16 22:44:30,197 : Genmon <xxx@xxx.com>
2020-07-16 22:44:31,552 : Error SMTP sendmail: {'': (501, 'Invalid command or cannot parse to address')} : mymail.py:438
2020-07-16 22:44:31,661 : Error in SendMailThread, sendEmailDirectMIME failed,retrying
2020-07-16 22:46:31,670 : Genmon <xxx@xxx.com>
2020-07-16 22:46:32,538 : Error SMTP sendmail: {'': (501, 'Invalid command or cannot parse to address')} : mymail.py:438
2020-07-16 22:46:32,568 : Error in SendMailThread, sendEmailDirectMIME failed,retrying
2020-07-16 22:48:32,573 : Genmon <xxx@xxx.com>
2020-07-16 22:48:33,333 : Error SMTP sendmail: {'': (501, 'Invalid command or cannot parse to address')} : mymail.py:438
2020-07-16 22:48:33,363 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2020-07-16 22:50:33,369 : Genmon <xxx@xxx.com>
```

CharlesGodwin, 2020-07-24:
I may be out of line but could the problem be that the internet/ISP/ router box is not powered at the time the Pi is sending. the Pi is powered but the other box(es)??? I ask because I've had this problem.

jgyates, 2020-07-24:
@CharlesGodwin, that is certainly possible and quite likely. Email notices will go out immediately once an outage is detected. Most routers take a min or so to boot. The question may end up being how long does this router take to come online and functional.

jgyates, 2020-07-24:
I can try putting a setting in the mymail.conf file that allows you to extend the wait time if an error occurs. I will include this in the next release.

jgyates, 2020-07-25:
I checked in V1.14.18. This has a new setting on the Advanced page (double click the gear in the top right to get to the advanced page). The new setting is "Extend email retry". This will add the specified number of seconds to the delay between an email failure and retry. To add on additional minute put the value of 60 here, etc. Let me know if you have any questions.

nkm8, 2020-07-28:
@jgyates Perfect - thanks for all your help!

nkm8, 2020-11-16:
@jgyates I tested this feature and never got an email. I haven't checked the logs myself, but I shared them via the settings page.

jgyates, 2020-11-16:
When did you send your logs? I have not received them today. If you are using the latest version your can download your logs in an compressed file with this command:

     http://ipaddress:8000/cmg/get_logs

If you outbound email is not working you can post your logs here or part of the logs. The file we are concerned about is mymail.log. NOTE your logs may have your email address use for outbound email in it so you may want to edit that if you are posting it publicly.

nkm8, 2020-11-16:
@jgyates I sent the logs yesterday. Looks like it is stuck in a loop repeating this error:
'''
2020-11-16 03:44:34,531 : Genmon <XXXXXX>
2020-11-16 03:44:35,831 : Error SMTP sendmail: {'<>': (501, 'Invalid command or cannot parse to address')} : mymail.py:464
2020-11-16 03:44:35,932 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
'''

I just sent a test email via settings that was successful.

jgyates, 2020-11-16:
I have not received your logs, I am guessing because your email is not working. The logs are emailed.

 What version of the software are you using. I would upgrade to the latest if you are not running 1.15.11 I would upgrade as there have been a few minor updates to the mail support since we started this thread.

This is a guess, but it looks like your recipient address is invalid. You can check this from the Notifications page. Possibly try deleting your notifications and re-entering them. That is just a guess.

The message "501, 'Invalid command or cannot parse to address'" is from your mail server. The software is trying to resend this every couple of minutes. this is the loop you are seeing. If it fails to send it is retired every 2 minutes (plus the Extend email retry value).

nkm8, 2020-11-16:
@jgyates 
- I am already running the latest release. 
- The recipient address is properly configured
- Sending a test email via settings works as expected
- The 'Sender Address' in settings was blank - I updated this, but haven't re-tested yet

jgyates, 2020-11-16:
Sender Account can be blank, if the "Email Account" setting has a valid email address. This is dependant on the mail server's requirements, but that is generally how it works. I use gmail and leave Sender Address blank because my account username is an email os account name is the email address. These fields can also be the same if that does it for your mail server.

nkm8, 2020-11-17:
@jgyates Interesting - I set a value, just in case. I also tested 'info' logging to confirm that email notifications are being sent successfully. I'll need to wait for the next outage to determine if I see this specific issue again. Thanks for your help!

jgyates, 2020-11-17:
FYI, with the 'info' category, you should get two emails when genmon starts. One that tells you genmon has started followed by another with the initial stats.