jgyates/genmon #411: Generac 25Kw NG (QT02524GNSNA) H-100 connection help wanted

usmc124, 2020-07-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

usmc124, 2020-07-04:
I have a couple 11kw generators running with genmon with no issues.  However, i would like to monitor my 25kw generator.  I am not finding a serial connector (molex) port like the others i have used.  There is a serial port in the front of the H-100 panel.  Do i connect to that?  Is there a setup procedure specifically for the H-100?  I am only finding Nexus and Evolution.  I have all the hardware ready to go.  Thanks- help is greatly appreciated as this product has been a major life saver for me.

jgyates, 2020-07-04:
The H-100 is configurable so it is my understanding that you could use either the front or the back connector, depending on your settings of the controller. I believe most use the front. This page has info on the H-100 setup:

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100--and-G-Panel-Controllers

usmc124, 2020-07-04:
Awesome thanks. That was dumb on my part- i didnt hit the link that says "view 6 more" which is where Appendix G is located.  Ill give it a shot.

closed 2020-07-04 by jgyates