jgyates/genmon #412: Can't get email to work

cmanion1983, 2020-07-05:
Using the outgoing mail settings, I'm unable to send test emails. I've used many different email address (as sender) all with the correct passwords and port settings. I haven't used gmail, only yahoo and aol. The error I most commonly get is 

An error occured: 
Error sending test email : Connection unexpectedly closed

Please try again.

Any ideas?

jgyates, 2020-07-05:
This thread has some info on gmail security settings. 

https://github.com/jgyates/genmon/issues/351

Most email providers have the equivalent of this now (OAuth vs "less secure apps"). 

The short answer is that you have to change a setting on your account to allow external apps that do not use the OAuth protocol. These are considered "less secure apps" as they only use a username and password for authentication.

cmanion1983, 2020-07-05:
It doesn't seem like they do that anymore, now you select an "app" and gmail, yahoo, etc generates a password for the account. You can select other. I have already done this and get the same result.

jgyates, 2020-07-05:
You must have two factor authentication enabled. With gmail, if you have two factor authentication enabled then you can setup application specific passwords, however you can not have two factor authentication enabled and use less secure apps. what most people do is create a separate email account just for sending, then make sure two factor authentication is disabled and then you can enable "less secure apps"

jgyates, 2020-07-05:
I just validated this on two gmail accounts. Two factor must be off before you get the option to enable less secure apps.

cmanion1983, 2020-07-05:
I created another gmail account. Turned on the less secure apps option. Double and tripled checked my settings, now i get this 

  | An error occured: Error sending email: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8 https://support.google.com/mail/?p=BadCredentials p53sm2835565qtc.85 - gsmtp')Please try again.
-- | --

jgyates, 2020-07-05:
these are the settings I use for gmail:

[image]

In addition I have two factor authentication disabled and "use less secure apps" enabled.

cmanion1983, 2020-07-05:
Okay that fixed it!

closed 2020-07-05 by cmanion1983