jgyates/genmon #413: Temperature sensor glob is too specific

lmamakos, 2020-07-06:
\
### Expected Behavior

Missing DS18B20 temperature sensor

### Actual Behavior

### Steps to Reproduce (including precondition)

The glob pattern at https://github.com/jgyates/genmon/blob/998b6ac801a92e787525eac08fe9f47c0eefd016/gentemp.py#L156 is too specific.  For example, my sensors look like:
```
lrwxrwxrwx 1 root root 0 Jul  6 09:09 /sys/bus/w1/devices/28-00000bc56fa5 -> ../../../devices/w1_bus_master1/28-00000bc56fa5
lrwxrwxrwx 1 root root 0 Jul  6 09:09 /sys/bus/w1/devices/28-020a91770670 -> ../../../devices/w1_bus_master1/28-020a91770670
lrwxrwxrwx 1 root root 0 Jul  6 09:09 /sys/bus/w1/devices/3b-4c880b6ac367 -> ../../../devices/w1_bus_master1/3b-4c880b6ac367
lrwxrwxrwx 1 root root 0 Jul  6 09:09 /sys/bus/w1/devices/3b-4c880b6ac389 -> ../../../devices/w1_bus_master1/3b-4c880b6ac389
lrwxrwxrwx 1 root root 0 Jul  6 09:09 /sys/bus/w1/devices/3b-6d800cc77f01 -> ../../../devices/w1_bus_master1/3b-6d800cc77f01
```
e.g., the address starts with `02` for one of the DS18B20 devices.

### Screenshot or Pictures relating to the problem (if possible)

_Sensor missing from display_

### Your Environment

- Generator Model:  22KW air cooled
- Generator Registers: 
- Genmon Version: Version V1.14.16

The glob pattern should probably just be `28-*` similar to the thermocouple devices.  It might also be desirable to use the same paths, either `/sys/bus/w1/devices` or `/sys/bus/w1/devices/w1_bus_master1`, though the latter might be too specific and an issue if there ever happens to be more than one 1-wire bus configured.

jgyates, 2020-07-06:
I updated gentemp.py to reflect this change. I did not update the version number so just update your repository and you should be good. 

thanks for the feedback. Let me know if you find anything else.

jgyates, 2020-07-06:
If you have any other issues you can post it to this thread. I will leave it open for a few days just in case.

jgyates, 2020-07-13:
I am going to close this thread, but feel free to comment here if you have related questions / comments.

closed 2020-07-13 by jgyates

griffths, 2023-07-02:
Not sure if this issue has anything to do with my problem but it appears to be similar/same symptoms so I am posting here for assistance.

I have a HAT made by @skipfire that has been working without fault for a year or so.  During that time I modified it to allow me to connect three DS18B20 temp sensors.

For quite a while, all three were displaying temperatures that were in the ballpark of what I would expect, in other words, working as expected.

However, one day I noticed that one the three temperatures had stopped being displayed which I attributed to a failed sensor.

Fast forward to now.   I am in the process of performing maintenance on my genset and figured I would replace the failed sensor since I was digging around inside the housing anyway.

Apparently there is more to this than a failed sensor because the replacement sensor is not being displayed either. 

It appears that I can only see two sensors at once because I can swap them out so I know that all three are functional.

I should also say that I have the sensor HAT from @skipfire and have the same results.

Following the [gentemp.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentemppy-optional) section of the Wiki I only see two sensors enumerated so am wondering if there was a change made to the software at some point that inadvertently introduced a bug/limitation on the number of sensors which are detected.

[image]

jgyates, 2023-07-02:
According to the output in your screen capture the operating system is only detecting two devices. gentemp.py just reads the following to directories for temp sensors:

/sys/bus/w1/devices/28-*/w1_slave for DS18B20 sensors

and 

/sys/bus/w1/devices/3b-*/w1_slave for K type thermocouples 

To see the list of DS18B20s:

     ls -la /sys/bus/w1/devices/28-*

 gentemp does not have a  limit on the number of devices.