jgyates/genmon #414: [Feature Request] MQTT post for update to GenMon available

GlennGoddard, 2020-07-11:
I utilize MQTT to pass info to Home Assistant and it works very well.  I also have the GenMon page as an iframe in Home Assistant.  I do have the email notification for update available turned on, and 'most' times I update it when email is received, however, life happens and I don't always get to it or remember to check it on the "About" page.  If there was an MQTT subtopic stating that an update was available then I could push other notification methods about the update.

I think this is the best fit location:
generator\Monitor\Platform Status\GenMon Update Available = True or False

I've looked through the MQTT, and I don't see anything along those lines, if I am missing it please point me to the MQTT topic.

jgyates, 2020-07-11:
This has been added to V1.14.17. 

After the update see Monitor->Generator Monitor Stats->Update Available

I put it under "Generator Monitor Stats" as that is the place for info related to the monitor software. Platform Stats is more for the system status.

closed 2020-07-11 by jgyates

GlennGoddard, 2020-07-11:
Wow, you are quick.  Thanks for the prompt response.  Too bad I have to wait till the next version after this to test it.

jgyates, 2020-07-11:
If you really wanted to test now you could edit genmon.py line 37 and make the version an older version then restart.

FYI, the presence of an update is checked by both the javascript and the python code.  The javascript checks for software updates if the About page is loaded. The python code checks for updates once a day (starting 24 hours after the software is loaded). The python code updates the flag that is displayed and sends out the email notice so in the first 24 hours or less after a release is posted to github you could see an update on the About page even if you do not see the flag set on the Monitor page yet. The flag will be set within the 24 hour period after the release. This is done this way to keep the python code from checking for updates too often and generating needless web traffic.

GlennGoddard, 2020-07-11:
Thanks, I will do that after I add an update badge to Home Assistant.