jgyates/genmon #415: Loading generator monitor

jjmacneil, 2020-07-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Login and see the GenMon website

### Actual Behavior

All I get is loading generator monitor....

### Steps to Reproduce (including precondition)

I’ve attempted to reboot the Pi with same results.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 22kw
- Generator Registers: Submit via the About page
- Genmon Version: V1.14.16

jgyates, 2020-07-12:
Is your serial communications connected? If serial is no connected or not working it may take several minutes before the web interface will come up. This is due to the web interface not knowing what type of controller is present. You can try refreshing the browser and web interface should come up once genmon has attempted to read all registers.

jgyates, 2020-07-12:
You can also use the ClientInterface.py program to access genmon. this bypasses the web interface and can give you insights into any possible problems. Info on that program is here: https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy

Once you run this program you can send the command

     generator: monitor

this will display serial stats.

jjmacneil, 2020-07-12:
Thanks for the quick response. Yes, the serial communication is working fine and this issue appears to have started after I updated to V1.14.16. Is there a known bug in V1.14.16?

jgyates, 2020-07-12:
Oh, if this is a system that was working then your problem could be that one of your conf files is corrupt. I am not exactly sure what this happens but I have seen it on rare occasion. If you can ssh into your system then you can resolve this by doing the following:

      cd genmon 
      sudo cp ./conf/genmon.conf /etc/genmon/

This will the main genmon conf file. Any settings on the main settings page will need to be setup again. Your email settings are in another conf file so those are retained. You Add On settings should also be retained.

If that does not work then you can try to replace the genloader.conf file
    
     sudo cp ./conf/genloader.conf /etc/genmon/

This file controls the load order and which add ons are enabled.

My best guess at why this happens (I have added code to mitigate this but it can still occur) is that if the SD card responds slowly and the restart attempts to modify the conf file while before the writes from the previous instance have not been fully saved, even though the old process has exited, but this is just a guess since it happens so infrequently.

jjmacneil, 2020-07-12:
Thank you. Do I need to restart the Pi after replacing the files? I ran the first set of commands via SSH, but did not restart the Pi and I’m still not able to login.

jgyates, 2020-07-12:
You can restart genmon, or reboot. To restart the software type this from the genmon directory:

    ./startgenmon.sh restart

jjmacneil, 2020-07-12:
Still spinning loading generator monitor after the first set of commands and after restarting GenMon I got my usual notification messages reporting the GenMon restarting so notifications are working but I still can’t see the GenMon website to manage anything.

jjmacneil, 2020-07-12:
How can I manually update GenMon via CLI without losing any custom configuration? I have an external WiFi adapter and last time I had to manually update I lost access to the Pi because the update ended up overwriting the config files.

Thank you

jjmacneil, 2020-07-12:
Just wanted to update you and let you know that I got it working. I also updated to the latest release as well.

Thanks for your help!

closed 2020-07-12 by jgyates

kspearrin, 2020-07-30:
@jjmacneil Can you share what the issue was?