jgyates/genmon #417: Linux Firewall Blocking Communication to GenMon

pridesleap, 2020-07-27:
Sytem Cofiguration:
    Non-Pi based System
    MODBUS to TCP Gateway at Generator
    Ubuntu Desktop Based Web Server

Sytem was configured and running. Established communication with the MODBUS to TCP Gateway configured in server mode and GenMon was displaying live data.

I enabled UFW and communication between the the web server and MODBUS gateway dropped off and eventualy completly failed.

I configured UFW with a rule that allowed all users on the local network to access port 8000 on the web server. 
`sudo ufw allow proto tcp from 192.168.10.0/24 to any port 8000`

 I also configured UFW with a rule that allowed all users on the local network to access port 23 on the web server (port used by the web server and MMODBUS gateway.
`sudo ufw allow proto tcp from 192.168.10.0/24 to any port 23`

I disable the firewall, everything comes back. I re-enable it and it fails. The rules I configured are inbound, do I need outbound rules to the MODBUS Gateway too? What am I missing here?

jgyates, 2020-07-27:
port 8000 is for the internal genmon web server
port 23 is for your modbus gateway

genmon uses port 9082 for internal communications between genmon.py and other apps like genserv.py. You can changed this on the advanced settings page

pridesleap, 2020-07-29:
Yep, that was it. I needed to add port 9082 to the firewall. All is working perectly. Thank you.

closed 2020-07-29 by pridesleap