jgyates/genmon #418: diesel gen fuel consumption

ewouk61, 2020-07-29:
would it be possible to get the gen mon to show fuel consumption and estimated time till empty? attached is part of the excel sheet that i use monthly for logging my generator inspections. 
in the variables i cheated them towards being conservative, but it should be within +/- 2hours. 
[to gen mon.xlsx](https://github.com/jgyates/genmon/files/4994454/to.gen.mon.xlsx)

jgyates, 2020-07-29:
Fuel consumption is already displayed for Evolution controllers. See this page:

https://github.com/jgyates/genmon/wiki/Appendix-H-Monitoring-Fuel-and-Power-Usage

When I implemented this I opted not to show time remaining as that is very dependant on the load of of the generator. If I were to implement this I would need to pick some random value for load (e.g. %50).

Also, depending on your model there is a fuel level sensor that is readable by the software (Evolution Liquid Cooled Diesel units should have this, mine does). If that is not already enabled, you can enable it from the Settings page (Use Sensor for Fuel Gauge) but that should only show up if you have an Evolution Liquid Cooled unit.

Let me know if you have any other questions.

ewouk61, 2020-07-29:
sorry. I should have been more clear. for H-panel gens

jgyates, 2020-07-29:
Part of the problem that I have with implementing this functionality for the industrial controllers is the lack of data. I don't have a reliable method to detect the fuel type. Also I do not have a method of detecting the engine size or rated kW output which I would use to correlate to a predefined fuel consumption. Even starting with a list of know models is something that I would have to track down. Having said all of that, this is do-able but it would heavily rely on input from the user to specific the model, fuel type, engine size, etc. I can put this on the list, but it is quite an effort in just running down all the data for each variant. I would guess that unless someone offered to help, this would not be considered until the Evolution 2.0 issues are resolved.

ewouk61, 2020-07-30:
"detecting the engine size or rated kW output which I would use to correlate to a predefined fuel consumption."
like a look up table? the excel sheet i attached, when the max kw, volts and amps per leg are entered the sheet calculates the current kw and load percent. then from there the sheet calculates what the run time in hours would be if the fuel tank is half full and the load is 50%, current load gph, and uses that to calculate total run time starting with a full tank and current load, and hours left on the tank at current load.
the only down side is where the gen is so lightly loaded that the engine is just basically idling, i don't know what the idle consumption is of the gen, the thing with the sheet is it dosnt care what the model of the gen or engine is just as long as it is diesel.
 this should only rely on the person to put the tank size, gen max kw.

jgyates, 2020-07-30:
The fuel consumption curve is different for each size engine and it varies based on the fuel type so a unique polynomial is needed for each engine variant. this is the most time consuming part of implementing (gathering the data).

Implementing this is not hard at all since I already have the mechanics in the code to do all of this. The legwork is the time consuming task (gathering the data for each engine variant supported and converting the load/fuel data into a polynomial).  You have this data for your specific generator model, but that would need to be gathered for each engine size and fuel type.

The evolution controller allows you to read the modbus registers to determine the model number (which gives you engine size and rated kw). I don't know how to get this data from HPanel and GPanel controllers so it would need to be input by the user.

liltux, 2020-07-30:
@ewouk61 I think you are over thinking this.  Most industrial installations worried about fuel will calculate @ full load conditions fuel consumption of their engine and how long they require to run.  They then make sure they never go below that level of fuel in the tank by configuring the user outputs of the industrial controller to alarm at that level for refuel.  Just a suggestion.  @jgyates I can try and help with data points for different engine configurations, but there maybe a lot of variables.

ewouk61, 2020-07-30:
if you use the gph/kw rule of thumb the engine size shouldn't be needed. 
where i work the only time full load run time is considered is in the planing stages of deciding on the gen tank, and the only time we see anything close to full load is when the gens are load banked. the rest of the time the gens are running down in the 20% load range.
the low fuel alarm only works if it has been set up. the guys who do startups in my area dont normally set the alarm up unless you are there at commissioning.

jgyates, 2020-07-30:
The way this is implemented for Evolution is that for a given genset generac publishes fuel usage. They typically publish values based on the engine size and fuel type and give at least 2 sometimes 3 data points (e.g. fuel rate at 50% load and fuel rate at 75% load). These data points are plotted and a polynomial for the curve or line is created. When the generator is running the percent load over a period of time using the power log. This value, along with the time period of the calculation is used to determine how much fuel is used over the time period using the polynomial for the genset. This is applied to the amount of fuel initially put into the tank over the time period to estimate the amount of fuel left in the tank.

To determine the amount of fuel left I would need to determine when the tank level is zero given an estimated load.

Maybe I am missing something, but it sounds like you are asking me to skip the fuel estimation and power log history and just use the load data that you are using.

ewouk61, 2020-07-30:
by using the rule of thumb you dont have to rely on being able to find the spec sheet for the gen, and it's data points.
i only have: nexus, H, and bi-fuel panels(will have to look and see if i have a G panel). the gen with the bi-fuel panel is from 2007, and the oldest H- panel is from 2009. i dont think i could find a spec sheet for them without calling generac.
the more and more we talk about this the more i am convinced this a simpler way, because it only relies on a person to put in the tank size, the gen kw, and selecting the correct voltage. (the one i sent is only for 3 phase, i will need to send the one for 120/240v)
the power log history would be useful for smoothing out the hours to empty(for when a hvac kicks in and out) but isn't necessary.

jgyates, 2020-07-30:
In general I like the idea of displaying the run time left in the tank however I do have some reservations. Also, I do not  doubt that they way you describe is simpler to implement. My concern are related to accuracy, scalability and support. The method you are suggesting relies on a unique constant that I do not have any info on how it was calculated so I would be unable to defend it when someone ended up asking "why does not this not work for me". Without more data to backup the rule of thumb constant I would not feel confident implementing this approach so at minimum I would need to see more data so I could determine that it would scale to generators that range in size from 25kw to 500kw (the range of diesel gensets supported). Also, this appears to only apply the diesel. Many would ask for this for LP as well.

CharlesGodwin, 2020-07-30:
Has anyone considered using a fuel flow meter/software. If you set a starting fuel level, measure consumption, then, based on current flow you can predict, arithmetically, time to empty. If you have a fuel level gauge you semi automate the 'how full was the tank' question.

ewouk61, 2020-07-30:
i originally came across the rule of thumb from the back of a generator service guys business card. but have found it in a couple of instances (https://www.americasgenerators.com/blog/post/2014/12/22/Generator-Fuel-Choosing-between-Diesel-and-Natural-Gas.aspx) but we all know how mis-information gets spread around the internet, so not believing me is wise. 

my gens range from 18-750kw(not all of them generac), with fuel tanks ranging from 40-2000 gallons.
my diesel generac gens have a analog sender on their rochester fuel guage. (6540 or 6543) https://rochestergauges.com/product/6500/
 with the Senior TwinSite Sender

for the ones with LP they might be able to pull the face off there gauge and replace it with the sender ( from data sheet "Rochester gauges may be easily converted for remote output by simply installing the R3D replacement dial and plug-in Hall Effect sender module." ) https://rochestergauges.com/product/m6200/

i have considered using a flow gauge/meter... until i saw the prices and the flows ( they ether only read high flow or low flow from what i found in the industrial plc world, $500 to 1000 each and i am self funding the 2 test units that i am testing for work, so that is not going to happen

grsthegreat, 2020-07-31:
https://fuelminder.biz/fuel_minder_store_products.html

cant you install a wifi compatible fuel level gauge and tie it in like the propane tank monitor?

ewouk61, 2020-07-31:
> 
> 
> https://fuelminder.biz/fuel_minder_store_products.html
> 
> cant you install a wifi compatible fuel level gauge and tie it in like the propane tank monitor?

the industrial gens already have a sender on the fuel gauge and the gen shows tank level as a % on the gen control panel

jgyates, 2020-07-31:
@ewouk61 

In looking into this, I found an issue with the way the fuel consumption is calculated for Evolution controllers. Due to this I will be making some changes to how this is calculated. I anticipate making a change that will accommodate industrial controllers as well. The current plan is to offer a method to enter the fuel consumption at 50% and 100% for a given industrial generator and the software will calculate the usage function. This will allow anyone with an H Panel or G Panel to have this feature as long as they have the data sheet for their generator. I will also try to add the estimated fuel duration till empty based on user definable percent load.

jgyates, 2020-08-02:
@ewouk61 

I have checked in V1.14.21 which corrects the fuel consumption calculation for Evolution controllers and adds the ability to calculate fuel consumption for industrial controllers. This page describes that you need to do to support fuel consumption:

https://github.com/jgyates/genmon/wiki/Appendix-H-Monitoring-Fuel-and-Power-Usage

Let me know if you have any questions.

jgyates, 2020-08-11:
I am closing this issue however post to this thread if you have any questions on this topic.

closed 2020-08-11 by jgyates