jgyates/genmon #419: Email Configuration - Send Notifications to Other Recipient?

pridesleap, 2020-07-29:
Searched through past issue posts, couldnt find this exact question asked.

I have configured GenMon email settings to send notifications via Gmail SMTP server. So far I have only been able to get GenMon to send the email to the same Gmail account that I am using to send the email from GenMon. In otherwords, I can only send it to myself. I'd like to send it to a seperate email address or even my mobile carrier smtp interpreter (###-###-####@vtext.com) Here is a screencap from my settings page:

[image]

jgyates, 2020-07-29:
The recipient email address is on the Notification page. Have you filled in the recipient info?

Also, if you want to send shorter messages to an email to text message mobile you can use the "Mobile Carrier Email to SMS" add on. You can also use the Notification page for email to SMS however it the email may get divided into multiple SMSs as some of the emails are too long, hence the Add on program.

If you have already tried this let me know.

pridesleap, 2020-07-29:
No, I did not fill out the recipient info on the Notifications page. I need to do a better job at reading the instructions. This did it, thak you.

Yes, configuring the Mobile Carrier Email to SMS option is on my list next and is probably going to be wehre I end up as far as mobile notifications go. I'm working through trying to get the basic features functioning first.

closed 2020-07-29 by jgyates