jgyates/genmon #421: Generic or GENERAC

pridesleap, 2020-08-02:
Not sure if this is a bug or a feature. I have been experiencing some strange behavior from what I suspect are serial comunication issues. I noticed this today ( I dont think its related) and wanted to ask the question.

Is this supposed to say GENERAC or Generic? I ask as I recieved emails form GenMon stating that it doesn't recognize the model of my controller. Again, I think its com issues but I didnt know if the "Generic" was something GenMon displayed when it couldn't identify the model or if it was just a typo.

[image]

jgyates, 2020-08-02:
The spelling was intentional. This means that the software can not perform an internet lookup from the serial number to determine your model number (for example G0070391). It means the software can detect it is an air cooled evolution based generator, bit does not know the model.

closed 2020-08-02 by jgyates

pridesleap, 2020-08-02:
Ok. Genmon is able to get a valid serial number from the controller though. How does it perforn the search to determine the model? I am trying to troubleshoot why it cant do this but dont know how that process works. I started getting some strange behavior yesterday after several days of error free operation and have been trying to troubleshoot that as well. I'm starting to think that these issues may be related. Im planning on submitting a seperate Issue regarding the serial comm problem.

jgyates, 2020-08-02:
I would look at /var/log/genmon.log. I should have more details. The lookup is based on this page:  https://www.generac.com/service-support/product-support-lookup

pridesleap, 2020-08-02:
I manualy did a lookup using my serial number and the GENERAC website and it worked fine. So it doesnt appear to be a problem on generac's end

I did look at the genmon.log file. There is a bunch of stuff in there from yesterday. I did several reboots in an attempt to try an diagnose my problem. I updated genmon to the newest version too. One of the last entries for yesterday was at 11PM last night. Genmon tried to do an internet search but it looks like there might have been a parsing error? I've attached the log file:

[genmon.log](https://github.com/jgyates/genmon/files/5012737/genmon.log)

jgyates, 2020-08-02:
This internet lookup is only done if 1) There is no valid model number and 2) The software version has been upgraded. 

It is possible that your internet was out or the pi was not connected to a network with the pi booted booted. If you want to force another check for the model number you can edit /etc/genmon/genmon.conf and remove this line:

     version = V1.14.20

Then restart genmon. This will force an update of the model number. 

Or you can upgrade to V1.14.21 as this will cause the lookup to occur.

pridesleap, 2020-08-03:
I should point out I am using a serial to TCP gateway with Ubuntu Linux on a PC.

Here is what I did:
I removed the version line /etc/genmon/genmon.conf, restarted genmon. No change
I updated genmon to V1.14.21, stopped and restarted genmon. No change.
I reset the serial to TCP gateway. No change.
Finaly I performed a reboot of the PC. No change.

The strange thing is everything was running fine until  yesterday. I had been checking the log files periodicaly to see if there were any issues but there was nothing like what Im running into now.

I have attached an updated log file:
[genmon.log](https://github.com/jgyates/genmon/files/5013358/genmon.log)

jgyates, 2020-08-03:
based on this log it looks like what may be happening is that your serial number is not read from the controller yet. Do you have any timeout errors on the monitor page under serial stats? The serial number is the first thing read from the controller, but if there is a timeout on that transaction it will continue to the next modbus registers requests.

jgyates, 2020-08-03:
also, if your wifi signal strength is marginal this could happen.

pridesleap, 2020-08-03:
Connection is wired from the generator through to the PC (through a switch and a router).

Here is a cap of the monitor screen:

[image]

pridesleap, 2020-08-03:
I have an updated log file. I think the monitor stopped and restarted itself:

[genmon.log](https://github.com/jgyates/genmon/files/5013484/genmon.log)

jgyates, 2020-08-03:
I would need to see the rest of the log files, you can send them from the about page, for example the file /var/log/mymodbus.log would tell what register timed out. If that singel timeout is for register 01f4 then this is your problem. Is the serial / TCP server up before genmon is started?

pridesleap, 2020-08-03:
Just sent the log files over. Yes, the server is up before genmon was started. The maintenance page is showing the valid serial number though. Wheres it coming from?

jgyates, 2020-08-03:
I loaded up your registers from the log file email you sent and it is working for me. My emulator also uses serial over TCP. Keep in mind that if you get a timeout on reading the serial number, the model lookup will fail. In subsequent loads the reading of the serial number can pass, but the model lookup will not occur unless the a software update is detected or the version info in the conf file is missing. With the current version of the software the serial number is only read once so I am guessing that what happened was that you restarted after a software update and the first registers read timed out. Both the logs mymodbus.log and myserialtcp.log support this detail. 

Your serial over TCP device may need to rebooted. These errors occured in the past 2 days, notice "Connection reset by peer":

2020-08-01 16:52:28,248 : Resetting SerialTCPDevice:ReadThread Error: serialTCP:'NoneType' object is not iterable : myserialtcp.py:105
2020-08-01 17:18:25,063 : Error in SerialTCPDevice:Read socket error: [Errno 104] Connection reset by peer : myserialtcp.py:165
2020-08-01 17:18:25,063 : Resetting SerialTCPDevice:ReadThread Error: serialTCP:'NoneType' object is not iterable : myserialtcp.py:105
2020-08-02 20:35:32,468 : Error in SerialTCPDevice:Read socket error: [Errno 104] Connection reset by peer : myserialtcp.py:165
2020-08-02 20:35:32,494 : Resetting SerialTCPDevice:ReadThread Error: serialTCP:'NoneType' object is not iterable : myserialtcp.py:105
2020-08-02 20:49:52,914 : Error in SerialTCPDevice:Read socket error: [Errno 104] Connection reset by peer : myserialtcp.py:165
2020-08-02 20:49:52,914 : Resetting SerialTCPDevice:ReadThread Error: serialTCP:'NoneType' object is not iterable : myserialtcp.py:105

This indicates the TCP connection was dropped due the serial to TCP device closing the connection. have you tried restarting your serial over TCP server?

I can make the software attempt another read of the serial number just before the lookup if it does not have a valid serial number. That may lessen the likelihood of a single dropped packet causing this issue.

jgyates, 2020-08-03:
I just checked in a update that will re-read the serial number if it does not have a valid serial number before the lookup. I did not update the version number as this is a minor change. If you update your software and remove the version= line in /etc/genmon.conf and save the file, then update the software from the about page then see if that fixes the issue.

pridesleap, 2020-08-03:
Actualy, I did try to remotely reset the TCP Server tonight a couple of times. Those times would corespond to the time stamps in the log file. Remotely I can initiate a reset of the Serial TCP server but not a full power cycle reboot. I just issued a reset of the TCP server and it produced the exact same errors in the myserialTCP.log file

pridesleap, 2020-08-03:
Ok, I just tried your suggestion.

- I updated the software via the ABOUT page.
- I removed the version line from the genmon.conf file
- and then updated the software from the ABOUT page again.

No change.

pridesleap, 2020-08-03:
FYI the mymodbus.log file is showing:
 **Error Receiving slave packet for register 01f4**

This is after a reboot of the TCP server

jgyates, 2020-08-03:
you could try to add another second or two to the modbus timeout on the advanced page. Does the serial number show up on the Maintenance page? It is odd that is the first serial TCP request that is experiencing a timeout.

pridesleap, 2020-08-03:
Yes the correct serial number does show up on the maintenance page. Is this value cached anywhere and the monitor is just reporting the cached value?

[image]

I added two seconds to the timeout, saved the change.
I removed the version line form the genmon.conf file
Stopped and restarted genmon.
No change

The monitor status page was showing a timeout upon the restart.

pridesleap, 2020-08-03:
Its weird that it is dropping that one register, every time. I think im going to call it for tonight. I want to connect directly to the TCP server and see what its doing before I take more of your time. I have to physicaly be there to use the configuration app to do it though as the app wont work through the VPN. Thanks for the help and your time with this, its most appreciated.

pridesleap, 2020-08-03:
I compared the Register value to the Maintenance Page. There is a differance between the two. Here are the caps:

[image]

[image]

So the register is giving me some oddball value (probably because the monitor cant read it). However the maintenance page is stil reporting the correct serial number.

jgyates, 2020-08-03:
that is normal. The register is formatted in a way that you have to remove every other value (i.e. only the low nybble of each byte). The serial number as displayed works on my system and when you look it up online.

pridesleap, 2020-08-03:
The serial number as displayed on the Register page? If so then the monitor is reading a valid serial number from the controller right? As I understand the problem, for some reason genmon isnt reading the serial number correctly (not recieving register 01f4). Even though the correct SN is displayed on the Maintenance page and apparently the Register page as well.

I'm heading over to the house tonight to do some troubleshooting. Any suggestions on where I should start? My plan is to use modscan to try and find where the register is getting dropped in the system. I also want to use modsim to try and force the register through to genmon.

jgyates, 2020-08-03:
Yes and no. Here is the problem as I see it.

In the version of the software before the update I posted last night the problem occured like this:

genmon starts, the first register read is the serial number registers. For unknown reasons this TCP request timed out. If this timeout occurs after a software update then the internet lookup will fail as there is no serial number. The lookup will not occur again until the version number in genmon.conf changes or is removed.

The new version that I checked in last night still reads the serial number first, however I added a new check that will re-read the serial number if it failed to get a valid serial number since the software was loaded. This version still only checks the model number on the internet if the version of the software has changed.

If you are not removing the version number after each restart of genmon, then the look up will not occur. to me the problem looks like on, occasion your first packet read fails due to a timeout. If this failure occurs on the restart after a software update then your model lookup will fail.

Once you are physically in the presence of the equimpent, I would restart the serial to TCP server, wait until it restarts, remove the version line in genmon.conf and reboot the system running genmon. If it still fails, send me the logs.

pridesleap, 2020-08-03:
Will do. I hadnt made any changes to the the system before I started seeing problems on Saturday. There was a step change and things have gotten progressively worse since then. Like I said, right now the monitor isnt coming up at all and there is 100% Timeout Percent Errors. I'll update you on my progress.

pridesleap, 2020-08-04:
Did some troubleshooting this evening. Bear in mind I had a new problem where the monitor wasnt recieving any datat from the TCP server. So I had to first troubleshoot that problem. Here is what I did:

**Troubleshooting total loss of communications with TCP Server.**
1. Performed a hard shutdown and reboot of the linux server. No change
2. Performed a power cycle of the TCP Server. No change
3. Rebooted the network switch where ethernet drop from generator terminates. No change.
4. Rebooted the main router. No change.
5. Reviewed the TCP and UART settings in TCP Server. Discovered the UART settings had chaged baud rate and parity to    incompatible settings. I restored these to the proper settings. No change.
6. Discovered the genmon.conf file Enable Serial TCP setting had been reset back to FALSE. Set this back to True.
7. Stoped and restarted monitor, communication with TCP Server re-established. Still had same issue with "Generic Air Cooled" being displayed on Maintenance page.

**Troubleshooting "Generic" Issue.**

1. Observed that the Monitor status page was reporting 0 Timeout Errors whereas before it was reporting 1.
2. I modified the genmon.conf file to remove the verison line
3. stopped and restarted genmon. No change.

I am sending the log files over now. I made sure to do a clean boot-up through to the start of the monitor so as to have a good set of log files to review. I restarted the linux sever just after 21:00 hrs EDT. log entries for this boot are time stamped at around 21:06. I am not seeing any Timeout errors this time in the logs.

However, the last entry in the  genmon.log file is giving the following error:
**_Error in LookUpSNInfo (parse request 1): cannot use a string pattern on a bytes-like object : generac_evolution.py:756_** 

Does this mean anything?

jgyates, 2020-08-04:
are you using python2.7 or 3.x?

pridesleap, 2020-08-04:
Python 3.x

jgyates, 2020-08-04:
This is a bug. I was not handling the lookup code correctly for python 3. I have duplicated this and checked in a fix. V1.14.22 should be available now. We were not getting there yet due to your serial over TCP issues, but I am glad we stayed with this as it found this bug. Thanks! let me know how your testing goes. Just update from the about page and when it reloads it should work.

pridesleap, 2020-08-04:
**FIXED!**

[image]

Yeah, I think I still jhave an issue with the TCP Server though. I dont like how it suddenly changed the UART settings for no reason. Who knows what else its deciding to do on its own. I am using a unit from StarTech because I already had it on hand. Do you have a recomendation for a converter that you prefer? I might replace this one.

No worries about working through this either. Part of the appeal of these projects is to flush out the problems. I used to be a field engineer for Siemens so it comes naturaly.

jgyates, 2020-08-04:
I use this one:

https://www.pusr.com/products/rs232/rs485-to-wifi-converters-usr-w610.html

Pros: you can power it from the 5V and GND on the controller and it does appear to work well
Cons: the web interface is a bit lacking and the manual is not well written

USR also makes Ethernet to serial (the one above is wifi to serial)

https://www.pusr.com/products/serial-to-ethernet-converters/

pridesleap, 2020-08-04:
Thanks for the info, im going to order one today

Also, I noticed this is my genmon.lof file this morning after a restart:

2020-08-04 12:06:52,717 : GenMon Loaded for site: 112 AVENUE ROAD
2020-08-04 12:06:52,775 : Validation Error: Invalid register length (base) 0000:33303033343135393730 2 10

Didnt know if it hanything to do with the bug you found last night.

jgyates, 2020-08-04:
This is likely due to a timeout error. The problem stems from the fact that the modbus protocol does not allow you to cancel a request and there is not official length to wait for a packet. If a request A is sent and the sender waits for x seconds and then decides the packet is not coming, then the sender sends request B but the remote end responds with packet A then a mis match can occur. The above validation error is the result of this occurring. Genmon checks the length of the returned data and if it is not valid it reports this error. The above error means that register 0000 was read and was expecting a length of 2 but got 33303033343135393730 (the contents of registers 1f4, the serial number register) with a length of 10. This tells me that your TCP/serial server is taking longer than the default modbus timeout to respond to this packet. I would add 2 or 3 seconds to the "Additional Modbus Timeout (sec)" parameter on the advanced page (double click the gear icon in the top right of the web page).

pridesleap, 2020-08-04:
I had previously added 2 seconds to the timeout but I just increased it to 5. If I continue to have problems I will either open a new cissue or (more likely) replace the TCP server. I honestly think its junk and is going to cause me pain in hte future anyway. Thanks again for the help.

jgyates, 2020-08-04:
Part of the issue here is that the "needed" timeout varies wildly once you add the serial over TCP based on the TCP server and network conditions. The current default timeout is the same for serial and serial over TCP. I am going to add an additional 2 seconds to the serial over TCP default timeout.

jgyates, 2020-08-04:
out of curiosity, once your system runs for a while, what is the percent timeout percent errors? I am using mine over a less than ideal wifi connection and add 2 seconds to the default and I have 3 additional seconds on the advanced page and my percent timeout was cut from around .85% to .50%. I am guessing that your percent errors is much lower or zero as you are using a wired connection.

pridesleap, 2020-08-04:
Funny enough I was just looking at that. It's currently at 0%. I had 1 Timeout error just after booting but the monitor had been running for a bit.

[image]

pridesleap, 2020-08-18:
So I wanted to give you an update on a couple of things I did over the last couple of weeks. I had been getting nusence alarms (loss of utility power, errent warnings about firmware revisions changing, etc..). My guess is that they were being caused by dropped packets but I couldnt tell for sure. I used ModBUS Poll to poll the TCP server and see If I could get similar timeout errors. Couldn't replicate the faults. I then replaced my Ubuntu Desktop PC with a PI4 running PI OS Lite. Im still using serial over TCP but the PI is functioning as the Genmon server. Now commissioned, I haven't seen the errant faults and alarm messages as  before. In fact, Genmon has been absolutely quiet. It looks like switching to the PI was the improvement.

jgyates, 2020-08-18:
Thanks for the details. that is good to know. I am guessing the overhead of the GUI is causing problems in the TCP responsiveness.

lwbeam, 2020-10-07:
I'm having the same problem getting GenMon to retrieve the model information for my generator. I have an air-cooled Pre-Nexus model, which was installed in 2010.
[image]

As the controller does not contain the machine's Serial Number, I've had to add it to the conf file. Although I've never had the controller replaced, it could have been before I took delivery of the unit. However, I suspect that Pre-Nexus controllers may never have been factory-configured with the Serial Number. I'd be interested to hear from other Pre-Nexus owners.

[image]

At first I thought it might be a problem with my slow internet connection or network setup (e.g., double-NAT). However, from within the python console, I've successsfully executing the following commands on the Raspberry Pi currently running GenMon. 

> from httplib import HTTPSConnection    
SerialNumber = "0005794169"
conn = HTTPSConnection("www.generac.com", 443, timeout=10)
conn.request("GET", "/GeneracCorporate/WebServices/GeneracSelfHelpWebService.asmx/GetSearchResults?query=" + SerialNumber, "", headers={"User-Agent": "Mozilla/4.0 (compatible; MSIE 5.01; Windows NT 5.0)"})
print(conn.getresponse().read())
conn.close()

The following is the information returned, which is similar to that for other/newer machines.

> \<?xml version="1.0" encoding="utf-8"?\>
<string xmlns="http://tempuri.org/">{"SerialNumber":{"Number":"0005794169","ModelNumber":"G0055231","Registered":true,"ValidSerialNumber":true},
"Results":[{"ObjectType":1,"Title":null,"Description":"G0055231","ObjectID":"56b3896e-1bce-4fa2-99ce-316d66380bec","ImageID":"3361eff0-7bca-4313-aefe-0f4112a30b50","ExactMatch":true,
"ImageId":null,"ImageLastModified":"\/Date(-62135596800000)\/","ImageUrl":"https://soa.generac.com/selfhelp/media//120000"}],"ManualModelNumber":null,"ResultsText":"1 Results"}</string>

I've also tried increasing the "timeout" to 30 seconds for the HTTPSConnection call, which has no effect. As a result, I'm fairly certain it's not a problem with my internet connection or network setup. Regardless of what I do, I still get the following appearing in the log file.

> 2020-10-05 11:44:08,048 : Generator Monitor Shutdown
2020-10-05 11:44:55,484 : New version detected: Old = V1.15.00, New = V1.15.00
2020-10-05 11:44:55,495 : Override Serial Number: <0005794169>
2020-10-05 11:44:55,908 : GenMon Loaded for site: Blue Point Cottage
2020-10-05 11:45:23,393 : Looking up model info on internet
2020-10-05 11:45:28,166 : WARNING: Initilization not complete after 30 seconds, possible communication failure. Check cabling.
2020-10-05 11:45:37,728 : Error in LookUpSNInfo (parse request 1): object of type 'NoneType' has no len() : generac_evolution.py:772
2020-10-05 17:18:38,838 : Error in ParsePreNexusLog length check
2020-10-05 17:18:39,521 : Error in ParsePreNexusLog length check

jgyates, 2020-10-07:
I don't have your register file to test with, but based on the data you posted the problem is likley this:

The current code only uses the Override serial number for looking up the model info if it appears the serial number read has been replaced. In every controller I have seen to date, if controller has been replaced the serial number is a "ff". The software currently checks the first 2 bytes of the serial number register, if these two bytes are both "f" then the assumption is that the controller has been replaced. In your controller, the format is different than what is expected. The first two bytes are "0" followed by two "F" values. If your outbound email is setup you can submit your logs (logs will send registers and logs, but at a minimum I will need the registers to validate a fix). This will allow me to test a fix that will likely fix your problem.

lwbeam, 2020-10-08:
I’ve sent my registers. Let me know if you didn’t get them. I’m waitIng for some good weather to swap off the controller with a spare I have. You might want to wait on making any changes until the new controller is in place, as that might solve to issue. No use making changing it for an oddball or failing controller.

jgyates, 2020-10-08:
I checked in V1.15.02. This will allow you to use the replacement serial number. The problem is that Pre-Nexus models do not support a serial number via modbus. The code was assuming that you would never use a SN on a pre-nexus. I modded the code to use a supplied serial number if it is present on a pre-nexus. Also, fyi, you can use the advanced page (double click the gear icon on the top right of the web interface) to enter the serial number.

lwbeam, 2020-11-12:
Thanks for making this fix. It's working perfectly now. I've finally had a chance to swap out the controller with a brand new spare. It also reports the SN as 00ff0000000000000000.

jgyates, 2020-11-12:
Great, I am glad it is working for you. Thanks for the feedback.