jgyates/genmon #422: GENMON System Warning - System Initializing not receiving data

boeingpilot, 2020-08-03:
System has been working fine for the past 8 days.  Of course, now that I'm 800 miles away, it's showing GENMON SYSTEM WARNING - System initializing not receiving data.

Tried rebooting pi; updating program, no success.  Realizing in reality I need to look at the physical hardware, anything I should try remotely?  Kind of concerned (tropical storm on the way).  

As anyone ever seen the line level converter quit?  Seems to me that's the only thing I could see inop if the PI is still responding.

jgyates, 2020-08-03:
I have had the level converter die when lightning hit nearby.  I also had a cable connection loose due to vibrations until I switched to the ATX style connector.

boeingpilot, 2020-08-03:
Well guess I’ll find out when I get home

boeingpilot, 2020-08-10:
ok - just got it apart....   When i opened the case, LEDs on Pi and level converter on (TX on the level converter).  Nothing obviously disconnected.  I did a hard power reset (disconnected and reconnected PoE, and voila, it works..  Concerning though....  Kind of negates the use of a monitor if the monitor drops out. Also, since I use the enhanced exercise, what happens if the Genmon starts the generator, and then I lose connectivity before it ends the exercise cycle?

That all said, I guess nothing to do but to wait and see.  I was in a bit of a rush this morning.  If it does this again, I'll first disconnect and reconnect the level converter.  If it comes back, safe to say it may be locking up.  If it doesn't, then it would point to an issue on the Pi (or at least the serial I/O.

boeingpilot, 2020-08-25:
Ok - system has been flawless since replacing the RS232 level converter.... until today.  So, one thing I had never done was remote start and transfer.  Wanted to make sure that worked.  Did it manually, no problem.  Ran the generator for 10 minutes and selected shutdown.  It went through its shutdown, but no change on Genmon.  Then noticed communications had stopped.  So, any suggestions?    Rebooting the pi via ssh did not fix.  Power cycled the whole Genmon (mine is powered PoE, so just unplugged and replugged at my switch) and it's back up.  But has me concerned.  One, I won't use automatic exercising with transfer (what if the communications drops after the start, then the Gen keeps on running?).  Also,want to make sure this is reliable.  

Here is how mine is assembled and where it is mounted..  

[image]
[image]

jgyates, 2020-08-25:
I would recommend powering your converter with 3.3V (pin 1) instead of 5V (pin2). 

I am a bit confused on the steps you describe above.

"Did it manually, no problem. Ran the generator for 10 minutes and selected shutdown. It went through its shutdown, but no change on Genmon."

Did you start by selecting Start and Transfer from the Maintenance page, or did you start it manually? Did you shut down from the generator or via genmon.

Also, I assume you are getting timeout errors, correct? or could it be that your config file is corrupt. Log files would answer that or you can look on the monitor page and settings page to make sure things look OK.

boeingpilot, 2020-08-26:
Ok will move power. Thought I had wires per diagram. 

Steps that I did 
- selected start generator and transfer from Genmon
- ran under load for 10 minute
- selected shutdown from Genmon
- generator shutdown (so it received the message) but a Genmon did not update.  When I went to monitor it showed no communications with controller. 
- rebooted pi via ssh, no change
- power cycled pi, Genmon back to normal. 

Hope that clarifies. I will move power lead tomorrow

jgyates, 2020-08-26:
3.3v is recommended. 5V will work, but the chip should have better tolerances with 3.3V. The diagram in the wiki shows 5V (i did not create the diagram so I can not edit it) but this note is on the page:

NOTE: If you RS-232 converter is powered using 5V the output of the receive lines to the PI are also 5V, according to the MAX3232 data sheet. The ideal solution would be to power your converter from the 3.3V pin on the GPIO header. I have yet to see a situation when powering your converter from the 5V supply did not work or damaged a pi, however the information on the PI and the MAX3232 chip does to point to the 3.3V supply being the ideal solution.

you may also want to validate your ground on the pi and the controller / battery have continuity. Basically the ground powering the 232, the controller and pi all need to be the same / connected. If you power from a source other than the battery or controller connector you must take care that all grounds are connected.

jgyates, 2020-09-02:
I am going to close this issue but feel free to comment in this thread if you have additional questions.

closed 2020-09-02 by jgyates