jgyates/genmon #424: Sync 2.0 stop command

Ricklws, 2020-08-09:
Hi I was wondering if anyone knows what the stop command actually does?  Will the stop command function if for say the generator is running and you get an alert that your propane is running and you still have no public power.  Does the command actual force a stop with the ATS active and keep it from running?

Just wondering before I go through the time to put this together.

jgyates, 2020-08-10:
If you issue the stop command when the generator is running with the transfer switch activated, the transfer switch will disengage and the generator will enter cool down mode (running for a few min without the load). Once cool down mode is complete the controller will stop the engine.

liltux, 2020-08-10:
@jgyates @Ricklws keep in mind the transfer switch can only go to utility position if utility voltage is present.  During a utility outage there is a proper procedure for shutting down the Generator in the Owners/Installation manuals that shipped with the Generator.

jgyates, 2020-08-11:
Closing this issue. If you have questions feel free to post here.

closed 2020-08-11 by jgyates