jgyates/genmon #425: DIY Propane Fuel Gauge Sensor

jgyates, 2020-08-11:
https://github.com/jgyates/genmon/wiki/Appendix-L-Adding-a-Propane-Fuel-Gauge-to-Genmon

curtis1757, 2020-09-01:
I plan on making this DIY gauge sensor but I have had a hard time finding a source to purchase the hall effect sensor that is linked to in the article....
Does anyone know where one of these Rochester 9704 modules can be purchased?
Thanks,
Curtis

zekyl314, 2020-09-01:
I found mine on eBay. It took quite a search. Sometimes I thought a about trying to source them myself and sell them too, but that could be difficult for me.

curtis1757, 2020-09-02:
Thank you Mike for responding.  Yes, I found some on EBay too for about $70-90.  But it is only a $15 part so I was hoping to get one around that price.  Rochester seems to only sell that part to companies like TankUtility.com that make propane tank monitoring systems.

I'll keep looking... if I find some at a good price I'll let you know too.

Thanks,
Curtis
  ----- Original Message ----- 
  From: Mike Walter 
  To: jgyates/genmon 
  Cc: Curtis ; Comment 
  Sent: Tuesday, September 01, 2020 4:01 AM
  Subject: Re: [jgyates/genmon] DIY Propane Fuel Gauge Sensor (#425)

  I found mine on eBay. It took quite a search. Sometimes I thought a about trying to source them myself and sell them too, but that could be difficult for me.

zekyl314, 2020-09-02:
I bought mine from eBay the seller name is controlit-ut, may you could try and contact them to see if they can get more?
-Mike

curtis1757, 2020-09-02:
What did you pay for it?
  ----- Original Message ----- 
  From: Mike Walter 
  To: jgyates/genmon 
  Cc: Curtis ; Comment 
  Sent: Wednesday, September 02, 2020 1:21 PM
  Subject: Re: [jgyates/genmon] DIY Propane Fuel Gauge Sensor (#425)

  I bought mine from eBay the seller name is controlit-ut, may you could try and contact them to see if they can get more?
  -Mike

zekyl314, 2020-09-02:
$22.99 but that was in 2017.  I had it sitting on my list of things to do and I finally had time and decided to work on this project and then proposed my idea to @jgyates to integrate into genmon.

curtis1757, 2020-09-02:
Thanks Mike for the info.... I'm working on getting genmon working with my generator now too.  That is great you got this project integrated in with genmon.

Curtis
  ----- Original Message ----- 
  From: Mike Walter 
  To: jgyates/genmon 
  Cc: Curtis ; Comment 
  Sent: Wednesday, September 02, 2020 1:43 PM
  Subject: Re: [jgyates/genmon] DIY Propane Fuel Gauge Sensor (#425)

  $22.99 but that was in 2017. I had it sitting on my list of things to do and I finally had time and decided to work on this project and then proposed my idea to @jgyates to integrate into genmon.

jgyates, 2020-09-29:
I am closing this issue for now but feel free to post related questions / comment in this thread.

closed 2020-09-29 by jgyates

curtis1757, 2020-09-30:
I decided to go all in and make my own hall-effect gauge sensor after finding the R3D sensor module is very hard to find, and if you do find one it is typically $80-$150.  I made one with about $24 parts (most expense item was the $13 for 25’ of wire!) 

 _Note: This sensor module I made is NOT compatible with the existing ones you can buy.  They have one direct output for the reading, this one I describe here has two, being the sine and cosine of the angle of the gauge dial.  Using the sin and cos values it is then easy to calculate the angle in some simple Python code)_

See  [Making R3D sensor module.pdf](https://github.com/jgyates/genmon/files/5302750/Making.R3D.sensor.module.pdf) for some details.

If anyone is interested in more details just email me, but when I get this all wrapped up I do plan to post more details and if jgyates is interested in the code to integrate into the DIY Propane Fuel Gauge Sensor, I can provide that or to anyone else that would like it.

I’d use it with the **genmon** package (which I got all wired up and connected correctly) but only to find I had version 1.11 of the Generac Firmware and therefore encrypted data protocol…. If that issue ever gets resolved I’ll use genmon and feed this gauge data into it… but for now will just use it standalone.

Curtis

jgyates, 2020-09-30:
@curtis1757 

Send me an email at generatormonitor.software@gmail.com and we can discuss alternatives to the Evolution 2 firmware issue.