jgyates/genmon #426: Gauge Replacement for 'Adding a Propane Fuel Gauge to Genmon'

jwtaylor310, 2020-08-13:
The post about custom-building a fuel gauge for LP tanks suggests that if the existing tank doesn't have a gauge that has a "Remote Ready Dial" it is necessary to empty the tank to replace it with one that has that feature.  Often that is not the case, and all that is needed is replacement of the dial assembly (not the whole gauge) which is easy to do and does NOT require emptying the tank.  Generac has published the procedure here: https://www.generac.com/GeneracCorporate/media/Library/content/all-products/parts-and-accessories/Tank%20Utility/Generac-Dial-Replacement-Guide.pdf   I used this guide to replace the dial on my LP tank and it only took a few minutes to install it.

zekyl314, 2020-08-13:
That is a great document, thanks for sharing.  I'm sure @jgyates will update the data I provided him.

jgyates, 2020-08-13:
Thanks! I updated the wiki. I had not seen this doc before.

closed 2020-08-13 by jgyates

lmamakos, 2020-08-13:
@jwtaylor310 - that's a really great document!  Where did you source the "remote read" replacement dial (and sensor, for that matter?)

zekyl314, 2020-08-13:
> @jwtaylor310 - that's a really great document! Where did you source the "remote read" replacement dial (and sensor, for that matter?)

Hey @lmamakos my LP tank had the dial already with the "remote ready" plug on it.  So I found what goes in there, links in the wiki page.  I then bought one from eBay I think.  No many people want to sell this for some reason.  I then worked on getting it connected to my RPI, and worked with @jgyates to get it implemented into his wonderful project.  

If you need a new gauge too, I'm guessing you'd want to look and measure your gauge like the document shows that @jwtaylor310 provided and then you could get the matching remote reader which is a simple 3 wire sensor.

zekyl314, 2020-08-13:
@lmamakos I looked on eBay just now, and I see a few for a lot more than I paid.  I only paid around $22 for my hall effect sensor, which is the Rochester one in the Wiki Page.

jwtaylor310, 2020-08-13:
lmamakos - I sourced the remote read dial from eBay.  My tank required part number 5909S02733 and a search on eBay currently lists several vendors offering that part for around $25.  Other tanks will require different part numbers as called out in the Generac write-up.  

I installed a commercial monitor which Generac re-brands and lists as part 7005.  It is fairly expensive - around $230 - but includes the sensor as well as the processing electronics with WiFi transmitter in a weatherproof case.  It runs off batteries and mine has operated for over a year without a problem even though it is mounted slightly below ground level in the "dome" of the underground tank.   You can find a listing for this unit on Amazon by searching for "Generac 7005".   Thanks to @jgyates, support for the 7005 is already included as an "Add-On" in Genmon.