jgyates/genmon #429: Outage log inop

jcherney, 2020-08-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Outage logs populated

### Actual Behavior

No recent outage logs

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

See attached pics

### Your Environment

- Generator Model: qto4842anax
- Generator Registers: Submit via the About page
- Genmon Version: 1.14.23
[image]
[image]

jgyates, 2020-08-17:
Do you have any errors in /var/log/genmon.log? Your SD card could be failing.

You can send you logs from the About page.

jcherney, 2020-08-17:
Haven’t checked those. I did send the logs to you.

jgyates, 2020-08-17:
I have not received any logs. Is your outbound email working?

jcherney, 2020-08-17:
Yes- I get the outage emails each time

jgyates, 2020-08-17:
I have your logs now. Looks like I introduced a bug in the last release that prevented outages from being logged on Evolution/Nexus. I checked in V1.14.24 which corrects this problem. Thanks for reporting this. Let me know if you have any other problems.

closed 2020-08-17 by jgyates