jgyates/genmon #430: Headless network change

Davidw4687, 2020-08-19:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

Davidw4687, 2020-08-19:
Without a monitor, keyboard and mouse attached I have no way to change the network association. Is some code available to detect the wireless network and handshake aka WPS?

jgyates, 2020-08-19:
I don't know of a WPS like feature for the pi.I typically setup my pi systems via Ethernet and ssh, then connect them via wifi one I have  established a wired connection.

You can configure wifi for a headless operation by updating this file via ssh:

https://www.raspberrypi.org/documentation/configuration/wireless/headless.md

This trick can be used to enable ssh before booting:

https://www.raspberrypi.org/documentation/remote-access/ssh/README.md

liltux, 2020-08-19:
@Davidw4687 if you can ssh into your pi you will want to edit your /etc/wpa_supplicant/wpa_supplicant.conf file.  I use the following command:
: ~ > sudo wpa_passphrase "NETWORK_NAME" "NETWORK_PASSPHRASE" | sudo tee \ 
-a /etc/wpa_supplicant/wpa_supplicant.conf
This will add your network info to the wpa_supplicant.conf file.  Then you can restart networking or (what I usually do) reboot the pi and it will connect to that network if possible.  I also add an extra network entry in the wpa_supplicant file for a network hotspot I use on my phone.  that way if the pi ever loose networking on the main AP I can always set my phone AP hotspot on to connect to the pi and recover.

liltux, 2020-08-19:
@Davidw4687 Re-reading your post, you could also right a bash script to run a iwconfig scan, parse the results, the connect to any open network.  But that would be ill advisable as someone could easily hi jack your pi with an open hotspot.

Davidw4687, 2020-08-20:
Thanks for the info. The issue comes up when the system has been setup and in place for some time, then you get a new router. This requires resetting up the Pi to associate it to a new network and password. I’m I missing something?

 

https://www.raspberrypi.org/documentation/configuration/wireless/headless.md

https://www.raspberrypi.org/documentation/remote-access/ssh/README.md

closed 2020-08-26 by jgyates