jgyates/genmon #431: External Tank Fuel Monitor Shows Zero after upgrade to V1.14.25

jwtaylor310, 2020-08-20:
I updated Genmon from V1.14.24 to V1.14.25 this morning.  I am using the External Tank Fuel Monitor Add-on.  The External Tank level now shows zero and about 10 minutes after the update was installed I received an email alert stating the level had dropped below 10%.   The tankutility.com web site shows a valid reading of 76% so it looks like the sensor is working normally.   Any chance a bug was introduced in the updated version?  Thanks!

jgyates, 2020-08-20:
I will need to see your log files to look into this.

jwtaylor310, 2020-08-20:
I have submitted the logs....

jgyates, 2020-08-20:
Nothing has changed from version 1.14.24 to 1.14.25 related to fuel monitoring. I do see errors in your log relating to tank utility login starting on the 18th. This may be due to a change in the login procedure / web format at tankutility.com. I checked in an update to the plugin that may workaround this. Update your software from the about page and let me know if the problem still exist. Note that I did not update the version number of genmon.

jwtaylor310, 2020-08-20:
That fixed it.  Thank you!

jgyates, 2020-08-20:
I will increase the version number to .26
Thanks for reporting this.

closed 2020-08-20 by jgyates