jgyates/genmon #433: Add ability or integration for multi factor authentication to enhance security for remote access to genmon

jjmacneil, 2020-09-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-09-10:
I will look into this and possibly add this to the todo list

pdeely, 2020-09-13:
Hello, I am a user of GenMon and quite willing to provide a "tip" on the Paypal link as incentive for this enhancement to be developed. My requirement (development suggestion) is for a login/password page to appear before allowing access to the GenMon monitoring page. I need this because i intend to set up port forwarding on my home router so that anywhere from a public Internet I can browse to the GenMon (TRaspberry PI) by entering: https://xxxpuIPaddressxxx:8081 whereby I designated port 8081 as the external port of the Raspberry PI which is accessed through internal browsing port 443. Please advise of your level of interest? Thanks and regards, Paul

jgyates, 2020-09-13:
@pdeely, simple username / password authentication is already supported via the "Optional - Webserver Security Settings" on the settings page.

Regarding two factor authentication. I will add this to the list of features to add.

jgyates, 2020-09-20:
MFA is now an option in V1.14.32.

If enabled it will required and authenticator application such as Google Authenticator or Authy to scan the QR code presented once the feature is enabled. MFA applies to both the normal and limited user login once enabled. Let me know if you have any issues or questions.

jjmacneil, 2020-09-20:
Awesome! Thank you so much! I really appreciate your quick response on this and your efforts for making it happen so quickly!

   -Josh

zekyl314, 2020-09-20:
Man, this app just keeps getting better and better!  Happy to support all your work @jgyates.

closed 2020-09-20 by jjmacneil