jgyates/genmon #434: MQTT Question generator/Status/Alarms/

liltux, 2020-09-11:
MQTT data missing or not getting formatted well.
This is on an H-100 controller.
Using MQTT to publish information to a subscribed service, I am trying to collect the actually current fault condition.  Looking through the code and MQTT data I have, I would suspect I should be able to get the information from generator/Status/Alarms/Alarm List.  But I get no data.
When I run MyClientInterface for generator:status_json, I see the Alarms/Alarm List/ and it has data.  The problem I think is the data is represented as a list and then runs through another round of place holders but no data.  Does this make sense?
[image]
[image]

Just curious if there is a nice work around to check if it is a list of dictionaries, if not then the list is treated as a list of values?

jgyates, 2020-09-11:
Can you send me registers for testing. My guess is that it is in the parsing in genmqtt.py

jgyates, 2020-09-12:
I checked in V1.14.30 that should correct this issue. Let me know if you have any issues.

jgyates, 2020-09-15:
I am closing this issue. Let me know if you have any problems with the fix I checked in.

closed 2020-09-15 by jgyates

liltux, 2020-09-15:
That worked!  Thanks.