jgyates/genmon #435: Power Zone Pro Digital Control Panel

tracerrx, 2020-09-11:
Any support for [Power Zone Pro Digital Control Panel](https://www.generac.com/Industrial/powerzone)

jgyates, 2020-09-11:
Not currently. Does it have serial outputs?

tracerrx, 2020-09-11:
Looks like it has Serial RS485/WiFi/Ethernet/BT

jgyates, 2020-09-12:
I can update the program /OtherApps/modbusdump.py to work over Ethernet. A quick web search implies that the Power Zone has modbus over TCP built in. We can test over RS485 or Ethernet. I will let you know once I have made the updates to test.

jgyates, 2020-09-12:
I am assuming you have access to a power zone controller to test, correct?

tracerrx, 2020-09-12:
I am currently replacing a genset, new one will have this controller... Aprox 2 months....

jgyates, 2020-09-12:
I updated the file /OtherApps/modbusdump.py to support modbus over TCP. This app (modbusdump.py) is used to probe either the serial or network interface for modbus response. Once you are in a position to test just let me know by responding on this thread. Since it will be about 2 months I will close this thread for now but we can still use it to communicate on this issue.

this doc implies that modbus over TCP should work: https://www.omnimetrix.net/wp-content/uploads/2020/03/IM-815-02-Generac-PowerZone-ModbusTCP-Configuration-Guide.pdf

We will need to know the ip address and the TCP port for the test.

closed 2020-09-12 by jgyates

jgyates, 2020-10-03:
FYI, I am tracking the feature request list here: https://github.com/jgyates/genmon/blob/master/feature_requests.md

jgyates, 2021-02-24:
FYI, I am in the process if adding support for the PowerZone controller and modbus TCP.  The existing software supports serial modus over TCP, however modbus TCP is a separate protocol.

Serial modbus over TCP is just taking the serial data as is and sending it over a transparent TCP passthrough 
Modbus over TCP removes the checksum (as it is not needed with  TCP) and adds a header to the packet.

Once you see version 1.6.0 checked in this will be the initial version that is ready for testing. I will post here once it is checked in however I wanted to give anyone interested an heads up this was coming.

jgyates, 2021-02-26:
Initial support for the PowerZone controller is now included in V1.16.00. This wiki page has been updated to include details on the powerzone configuration:

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

Currently the alarm and event log are not read by genmon. Hopefully that will be added soon if I can figure out how to do that.