jgyates/genmon #436: Unsupported Sensor 1 (0x0032) : 0.00

burr178, 2020-09-13:
great software man, i love it, i think that  Unsupported Sensor 1 (0x0032) : 0.00 may be temp of the gen head, or something  about temp, i look at the tech manual it doesn't show any thing about temp sensor,   do know man just try to help out. sorry if i dont know what talking about

jgyates, 2020-09-13:
on thing to consider is that the decimal place in the display is done by my code. That is just a guess that this value is displayed that way. The raw value read from the controller is not divided by 100 but the displayed value is divided by 100.

jgyates, 2020-09-15:
Closing this issue for now. If anyone has any input in this feel free to post to this thread.

closed 2020-09-15 by jgyates

liltux, 2020-09-15:
@burr178 what generator do you have?  Air Cooled or Liquid cooled?  I know there were some H-panel generators that I have ran into in the past that would have a reading or display for oil temp, but no oil temperature sender was actually installed on the engine, so it always read 0.  Just a thought.

burr178, 2020-09-15:
My is air cooled with Evo 2 controller. I get a value with the gen is
running. it was reading 95.0 to 105.0.

liltux, 2020-09-15:
Ok, so the analog values that I am aware of that are measured by the controller are:
-Current via CT1 and CT2 on wires 11 and 44 respectively.
-Voltage via the AVR sensing on wires 11s and 44s.
-N1 and N2 Utility voltage sensing.
-Ambient Temperature for the battery charger via a thermistor in the harness.
-Battery Charger current via the battery charging circuit and the controller, internal.

Is your unit 3 phase?

burr178, 2020-09-15:
No it 16kw single phase air-cooled, I just noticed it had a value that was
changing with gen running,  it would sit 95.0 and it would increase to
105.0 and that back done to 95.0. I will watch more closely next time it
runs, and see if it correlated to anything.  Just try to figure it out if
can.