jgyates/genmon #438: Are there APIs?

skipfire, 2020-09-18:
Is there an API for accessing the GenMon data?  I have a home automation system with text displays that shows non-desired states like open doors and windows, smart switches that are offline, low disk space on computers, car that is unlocked or needs to be charged, etc.  I would put my own service that translates in the middle, but is there an API to get the data already?

The kind of data I would be looking for would be:
Generator state (idle, running, fault)
Last quite mode test
Last transfer test
Load if running (kw)
Time running

jgyates, 2020-09-18:
Are there APIs? Yes. Are they documented? No, however they are pretty straight forward. You can call them a couple of ways. 

1) Use the python library myclient.py to send commands to genmon. These commands are in the same format as the command line interface, e.g.: 

        generator: status_json

2) You can make calls to the web server. For example if you pi is at IP address 192.168.1.10 you can make use this url:

     http://192.168.1.10:8000/cmd/status_json

The different commands are listed in the code at this location:

https://github.com/jgyates/genmon/blob/master/genmon.py#L482

Another command of interest would be the "getbase" command:

      http://192.168.1.10:8000/cmd/getbase

This is command is used to change the colors of the menu on the web interface. It returns one of the following states:

ALARM, SERVICEDUE,EXERCISING, RUNNING,RUNNING-MANUAL,OFF, or READY

You can get all the info that is displayed in the web app with these commands:

     status_json, maint_json, outate_json, logs_json, monitor_json

If you enter the url with your ip address in your browser you will see the output in JSON format. This can easily be used in most programming languages hat support sockets and JSON

skipfire, 2020-09-18:
Is there a path for authenticating with API?

lmamakos, 2020-09-18:
@skipfire you might also be able to use the MQTT protocol to integrate with genmon.  This is what I use with Home Assistant and MQTT "sensors" configured there.  genmon will push out data to a bunch of different topics, covering much of what you're looking for.

skipfire, 2020-09-18:
> @skipfire you might also be able to use the MQTT protocol to integrate with genmon. This is what I use with Home Assistant and MQTT "sensors" configured there. genmon will push out data to a bunch of different topics, covering much of what you're looking for.

@lmamakos I could, but I'd prefer not to as that would just add another layer that I'm not using for anything else and and would increase me to 4 layers involved in a single request.  I know I'm probably more of an edge case having built my own home control system instead of using other systems, but mine has been evolving since about 2001 and does what I want.

jgyates, 2020-09-18:
@skipfire 

How to login depends on your language. With python do this:

        import requests
        url = 'https://192.168.1.10'
        values = {'username': 'myusername',
                  'password': 'mypassword'}

        try:
            session = requests.Session()
            session.verify = False  # disable SSL checking for self signed keys
            r = session.post(url, data=values)
            #print r.text

            r = session.get(url + "/cmd/status_json")
            print r.text
        except Exception as e1:
            print ("Error: " + str(e1))

This should print the json data from the status page assuming you change the ip address, username and password above.

skipfire, 2020-09-18:
Thanks, I should be able to translate it from Python.

jgyates, 2020-09-18:
I updated the wiki entry for this: https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications

closed 2020-09-18 by jgyates

skipfire, 2020-09-29:
Thanks for the help, I was able to get things working with almost no trouble.  I may even start working on a mobile app.