jgyates/genmon #439: Feature request: Status Screen enhancements Utility Voltage Trend and weather

gzebrick, 2020-09-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

Thank you for your work on this project. I rely on the Generator Monitor for my home Honeywell/Generac generator.
I know the monitored variables have some built-in trending functions when viewed via the Monitor>Setup screen, but those trends are small and not consistent.

Would it be possible to add a full-time trend for the Utility Voltage identical to the kW Output trend that is viewable from the main Status screen? Same graphic object with hour/day/week/month trending as is already setup for kW? I know there's a min-max value displayed under the Line: section, but there's no time stamp on the values and there's no way to visually indicate if the min/max values are recurring or one-time events. A trend line would be much better.

I recognize that the amount of trend data could be excessive so they may be a need to limit the saved sample count and/or edit the voltage delta threshold (resolution) to prevent the trend data from being excessive, but it would be nice to have see a record of just how stable the utility feed is on an on-going basis. Perhaps the sample count max / trend change of value threshold could be adjustable parameters accessible via the setup screen.

I've also enabled the weather data report, it seems to work fine, but perhaps it could also be added to the home Status screen? Maybe make that an option and/or just show truncated data (Current Conditions only) - with the full report still under the Monitor screen?

Thanks' again. It's a fantastic bit of work you've done.

jgyates, 2020-09-20:
Let me think on this request. My concerns are the life of the SD card and the size of the log file if there is continuous data collection. I know some folks have implemented this type of graph in an external application via MQTT interface with external automation apps.

I will consider the weather also...

jgyates, 2020-09-29:
I am tracking the feature request list here: https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will be closing this thread for now but the request is still on the master request list.

closed 2020-09-29 by jgyates