jgyates/genmon #440: No web interface after upgrade to V1.14.32

UPS99, 2020-09-21:
FYI, actually.  After applying the upgrade, the web interface was no longer working.  Running "sudo python genserv.py &" produced this error: 

"This program requires the pyotp library. Please see the project documentation at https://github.com/jgyates/genmon.
Error: No module named pyotp"

I didn't see specific info about that, so I installed pyotp and it fixed the problem.

sudo apt update
sudo apt install python-pyotp

jgyates, 2020-09-21:
Your logs would be helpful. Are you running on a pi or another platform. Under normal conditions you would see this message in your /var/log/genloader.log file:

     Warning: required library pyotp not installed. Attempting to install....

And then this command would be issued:

      sudo pip install pyopt

If you are using python 3 then the pip would be replaced with pip3.

installing with apt is fine too. If there were errors on the install they would be logged in genloader.log also

UPS99, 2020-09-23:
I am running a standard pi.  genloader.log is attached.  It shows it tried to automatically install pyotp v2.3.0, but there were errors.

The python version is 2.7.13

 

It has been working fine with the apt install of pyopt.

jgyates, 2020-09-23:
Attachments are stripped when you reply via email. You will either need to attach the file via the GitHub web interface or, if your outgoing mail is setup you can submit logs to me via the button on the About page of the genmon web interface.

UPS99, 2020-09-23:
I do have email set up for alerts and paging.  So I just Submitted Logs.  Hopefully that works.

jgyates, 2020-09-23:
I got your logs. It looks like you have some issues on your pi. There are errors in your genloader.log that show this command is failing on your system:

     sudo pip install pyotp==2.3.0

For python 3 this would be the command:

      sudo pip3 install pyotp

If you want to try to correct this problem I would try what this thread recommends:

https://stackoverflow.com/questions/31719816/pip-fails-to-install-anything-error-invalid-command-egg-info

My guess is that your pip index files are corrupt, which may be a a sign of a failing SD card, but that is just a guess.

jgyates, 2020-09-25:
I am going to close this issue as I believe it is specific to your system. If you would like to troubleshoot this feel free to post info or questions to this thread.

closed 2020-09-25 by jgyates

UPS99, 2020-09-25:
I ran the command manually with added timeout and it succeeded without error:

sudo pip --default-timeout=1000 install pyotp==2.3.0

It might be a delay in the firewall routing I have?

jgyates, 2020-09-25:
that is possible or delays caused by your ISP. I have never seen this issue before but I can try adding this delay to the command when it is issued by genloader.py

jgyates, 2020-09-25:
Actually a better option would be to add this to /etc/pip.conf in the [global] section:

     timeout = 60

This assumes you have at least Rasbpian 9.

UPS99, 2020-09-25:
I found this fix on several forums when searching for pip install errors:

 

https://stackoverflow.com/questions/50305112/pip-install-timeout-issue

https://askubuntu.com/questions/905196/pip-fails-with-readtimeouterror

jgyates, 2020-09-25:
The above just adds the timeout you mention to the pip.conf file instead of the command line. If the delay is specific to your system this would add it for every pip command issued on your system. You can change the 60 value mentioned above to what every delay work. 1000 is probably a bit much (1000 seconds).

UPS99, 2020-09-25:
It definitely did not take more than a few seconds for the install to fully complete.  I assume that timeout allows for the worst case scenario if there is no response.  A setting as low as 10 seconds would probably be enough.