jgyates/genmon #441: Pi4 : Just a FYI

jmone1, 2020-09-28:
More of an FYI for those running up Genmon on a Pi4 as I'm not sure there is a "forum" as such.

I was running the Pi4 off the Generac Battery with a 12V to USB Socket (then a USB A to USB Cable).  Worked fine for weeks but I then started to see restarts (that got more and more frequent), and eventually my SD Card was corrupted.  Looking at "dmesg" output I was getting low power issues.  Time to upgrade the Power Supply.... but, 

It seems it is hard to get an all in one 12V Bare wire to USB C Cable/Adapter that will do 3amps.  I tried a few different combinations to supply enough power:
- 12v DC to 230v AC "Coffee Cup" inverter then the PI AC Adapter:  Bad idea all round + you lose the common ground
- 12v bare Wire to Cigarette Lighter Socket (10A fuse) --> 3A USB Car Adapter --> USB C Cable --> Pi4:  So far so good, no power issues and it is working on long cable runs.

I'll see how it runs over the next month or so, and if I've issues I'll report back.

jmone1, 2020-09-28:
One potential issue I do see on the Pi4 "dmesg" output is the following Python "Taint" that seems to be related to Genmon trying to grab throtted info from the Pi4:
------------------------
```
[   43.405782] ------------[ cut here ]------------
[   43.405799] WARNING: CPU: 3 PID: 671 at drivers/firmware/raspberrypi.c:212 get_throttled_show+0x6c/0x70
[   43.405804] deprecated, use hwmon sysfs instead
[   43.405809] Modules linked in: 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 cfg80211 bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp                              (C) v4l2_mem2mem bcm2835_mmal_vchiq(C) rfkill videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) raspberrypi_hwmon                               videobuf2_common snd_pcm videodev mc snd_timer vc_sm_cma(C) snd rpivid_mem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[   43.405877] CPU: 3 PID: 671 Comm: python Tainted: G         C        5.4.51-v7l+ #1333
[   43.405880] Hardware name: BCM2711
[   43.405883] Backtrace:
[   43.405893] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[   43.405898]  r6:e2214000 r5:00000000 r4:c129c8f8 r3:c3c52ad1
[   43.405906] [<c020d748>] (show_stack) from [<c0a39a44>] (dump_stack+0xe0/0x124)
[   43.405914] [<c0a39964>] (dump_stack) from [<c0221c70>] (__warn+0xec/0x104)
[   43.405919]  r8:000000d4 r7:00000009 r6:c0e2a724 r5:00000000 r4:e2215d8c r3:c3c52ad1
[   43.405925] [<c0221b84>] (__warn) from [<c0221d0c>] (warn_slowpath_fmt+0x84/0xc0)
[   43.405930]  r9:c0e2a724 r8:000000d4 r7:c08a31d0 r6:00000009 r5:c0e2a7bc r4:c1204f88
[   43.405936] [<c0221c8c>] (warn_slowpath_fmt) from [<c08a31d0>] (get_throttled_show+0x6c/0x70)
[   43.405941]  r9:e21fa000 r8:e2f0f280 r7:ef982210 r6:c0c460e0 r5:e21fa000 r4:ef97f2c0
[   43.405948] [<c08a3164>] (get_throttled_show) from [<c07599c4>] (dev_attr_show+0x28/0x58)
[   43.405952]  r5:c128e4fc r4:e34480b0
[   43.405960] [<c075999c>] (dev_attr_show) from [<c049714c>] (sysfs_kf_seq_show+0x90/0x108)
[   43.405964]  r5:00001000 r4:e34480b0
[   43.405971] [<c04970bc>] (sysfs_kf_seq_show) from [<c0495574>] (kernfs_seq_show+0x34/0x38)
[   43.405976]  r9:e2215f60 r8:00000000 r7:e2215f60 r6:00000001 r5:00000001 r4:e34480b0
[   43.405984] [<c0495540>] (kernfs_seq_show) from [<c042696c>] (seq_read+0x228/0x51c)
[   43.405989] [<c0426744>] (seq_read) from [<c0495bd8>] (kernfs_fop_read+0x134/0x1a4)
[   43.405995]  r10:00000003 r9:e2f0f280 r8:b3a057fc r7:e2215f60 r6:e2215f60 r5:00001000
[   43.405998]  r4:c1204f88
[   43.406006] [<c0495aa4>] (kernfs_fop_read) from [<c03fb184>] (__vfs_read+0x48/0x1d8)
[   43.406011]  r10:00000003 r9:00001000 r8:e2215f60 r7:e2215f60 r6:c3c52ad1 r5:e343ccc0
[   43.406014]  r4:c1204f88
[   43.406021] [<c03fb13c>] (__vfs_read) from [<c03fb3b0>] (vfs_read+0x9c/0x174)
[   43.406026]  r9:e2214000 r8:b3a057fc r7:e2215f60 r6:b3a057fc r5:e343ccc0 r4:00001000
[   43.406032] [<c03fb314>] (vfs_read) from [<c03fb768>] (ksys_read+0x70/0xec)
[   43.406037]  r8:b3a057fc r7:00001000 r6:e343ccc3 r5:c1204f88 r4:e343ccc0
[   43.406043] [<c03fb6f8>] (ksys_read) from [<c03fb7fc>] (sys_read+0x18/0x1c)
[   43.406048]  r8:c02011c4 r7:00000003 r6:00000010 r5:b3a057fc r4:00001000
[   43.406054] [<c03fb7e4>] (sys_read) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[   43.406058] Exception stack(0xe2215fa8 to 0xe2215ff0)
[   43.406063] 5fa0:                   00001000 b3a057fc 00000010 b3a057fc 00001000 00000000
[   43.406068] 5fc0: 00001000 b3a057fc 00000010 00000003 00001001 b6e716d8 00001001 000005e8
[   43.406072] 5fe0: 00000002 afbfa188 00000000 b6dee8bc
[   43.406076] ---[ end trace 61a606195d92ded9 ]---

```

jgyates, 2020-09-28:
Thanks @jmone1 . I did not realize that get throttled methods is deprecated. I will add this to the list of things to change. Keep us updated on the power issue on the pi4. I use a pi4 for testing but it is using serial of TCP so I am not powering like you are so you data is good feedback for others.

jmone1, 2020-09-28:
Thanks @jgyates.  It has been rock solid for a day now.

liltux, 2020-09-28:
[image]
FYI, I have used something similar to the above for well over a year in a few Generators.  I have yet had an issue with them except when I hooked up the DC backwards one time on the input. : )

jgyates, 2020-09-28:
@jmone1 , FYI V1.14.33 is checked in and it has the fix for the get throttled issue you mentioned.

jgyates, 2020-09-28:
Here is one with a USB-C plug.

https://smile.amazon.com/HOMREE-Step-Down-Converter-Module-Charger/dp/B07ZQB6S3L/ref=sr_1_5?dchild=1&keywords=12v+to+5v+dc+converter+3a+usbc&qid=1601328672&sr=8-5

closed 2020-09-28 by jgyates

jmone1, 2020-09-28:
Thanks!
- Tested V1.14.33, I can confirm that it is now fixed and working just fine!
- I did look at the HOMREE but it is not available in Australia and would take a month to arrive from Amazon US.  I did consider the Micro USB Version with a Micro to C adapter (or cable), or this one that has USB A plugs (then a USB A to C cable).... but it looks like you would need to splice some cables to reach the Battery Terminal (and they do look very light gauge cables).  Anyway, my current setup gives me about 3m of cable so I can position the Pi4 well away from the Generac.
[https://www.amazon.com.au/Direct-Converter-Supply-Female-Charger/dp/B01EFUHK08](url)