jgyates/genmon #442: Useful tips

jgyates, 2020-09-28:
I have created this wiki entry

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

to hold useful tips and tricks that are useful when using or setting up genmon. If you have any other tips you think are useful post them to this thread and I will add them to the wiki

skipfire, 2020-09-29:
Not sure if it follows what you are thinking, but I used a couple hardware pieces that gave me nice clean hardware.
For the serial I used a RS232 hat https://thepihut.com/products/modmypi-serial-hat-rs232 and then I used the HighPi case https://thepihut.com/products/highpi-raspberry-pi-4-model-b-case.  Then I added a standard serial cable so that I could put distance between the Pi and the port on the controller.

liltux, 2020-10-02:
@skipfire,  pishop.us is another distributor for those items, I use them as well ; )

jgyates, 2020-11-23:
I added a link to this thread on the Tips and Tricks wiki entry:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon

mefbgil, 2021-01-04:
Hi everyone - very exciting day for me - got Genmon running inside Docker connected to my Generac thru a USR-W610.  Everything worked on the first try - powered cable to the USR, USR setup on my WiFi, Dockerfile build, and TCP connectivity to the USR - all good - EXCEPT - MQTT.  Any ideas where I should look for issues?  The MQTT log is empty and no trace of error message in any other logs.  Any help is good help.  Thanks!

madisdev, 2021-01-04:
That’s good to hear. Please share your docker file. But from my experience I think you need to setup a docker network that both mqtt and genmon share.

mefbgil, 2021-01-04:
> That’s good to hear. Please share your docker file. But from my experience I think you need to setup a docker network that both mqtt and genmon share.
> […](#)

Files attached to my original comment.  Other MQTT clients connect to my MQTT server, no prob.

jgyates, 2021-01-04:
Since this is not really about useful tips, please create a new thread on this topic so it can be easily searched if others have issues. I will respond in that thread.

RubyBailey, 2021-01-31:
Lesson learned - Serial communication errors and grounding.
I could not get serial communication established after multiple rechecks of wiring and devices. All serial tests passed. Even used different devices (PI3+ and PI0W as well as different TTL to RS-232 converters) while troubleshooting. 
In [https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting](url) it say: "If you power from a source **other than the battery** you will need to connect a ground wire from the molex connector on the controller to the RS-232 9-pin connector. ... Bad ground connections will cause serial errors or even a lack of communication on the serial link. **To validate your ground has a good connection, use a multi-meter to measure continuity between the generator battery negative terminal and the ground on the the pi.**"
I used a 12Vdc (from battery) to 5Vdc power supply and verified that the battery negative terminal and the pi ground had continuity. Serial communication would not work.
Only after connecting the ground from the molex (pin 2) to the RS-232 connector did the communications start working.  
May be worthwhile putting a bit stronger language in the above referenced section. Would have helped me. :-) Thanks.

jgyates, 2021-01-31:
I updated the wiki with your suggestion. Thanks

sarahmva, 2021-05-26:
Not sure where to ask this. I have a residential Nexus controller and I do see a number of "Unsupported Sensors" - most of them show values when the generator is running. These are the sensors in standby.
Unsupported Sensors :
            Raw RPM Sensor : 0
            Hz (Calculated) : 0.0 Hz
            Unsupported Sensor 1 (0x0032) : 0
            Unsupported Sensor 2 (0x0033) : 160
            Unsupported Sensor 3 (0x0034) : 1

How can I figure out what these values mean? I think 0x0032 is Voltage of one leg of the generator.

liltux, 2021-05-26:
The Nexus and all air cooled controllers to my knowledge only record L-L voltage.  So it would be a sensed 240VAC for 60Hz or 220VAC for 50Hz.  There is no neutral reference wire bank to the controller.

jgyates, 2021-05-26:
The unsupported sensors are values that appear to change on some controllers, but I could not determine what the values represent by observation. I included the option to view this if someone wanted to take an educated guess.

skipfire, 2021-05-29:
Adding rare earth magnets to the bottom of a HighPi case works really well for placing the Pi on the sidewall of the generator. I have mine on the backside of the front in the compartment with the battery.  After much testing I found my Pi had the best WiFi signal when positioned as near to the side as possible and with the USB ports facing inward.  Just moving around and testing took me from -72dBm to -63dBm, a 3x improvement.
[image]

closed 2021-07-30 by jgyates