jgyates/genmon #444: SNMP Error in SnmpCallbackFunction: setComponentByType() got multiple values for keyword argument 'verifyConstraints' : gensnmp.py:475

s5y3XZpGvQPApqR, 2020-09-29:
### Expected Behavior

SNMP responses from gensnmp.py

### Actual Behavior

No response. Error in gensnmp.log

### Steps to Reproduce (including precondition)

I enabled SNMP, then while booting noticed it attempt to pip install various modules, some of them failed 
2020-09-29 13:01:38,848 : Error in SnmpCallbackFunction: setComponentByType() got multiple values for keyword argument 'verifyConstraints' : gensnmp.py:475
genloader.log:No matching distribution found for pysnmp
genloader.log:2020-09-29 12:38:39,585 : Warning: required library pysnmp not installed. Attempting to install....
genloader.log:2020-09-29 12:40:34,200 : Error in InstallLibrary using pip : pysnmp : Exception:

I manually installed pysnmp using apt

### Screenshot or Pictures relating to the problem (if possible)

### Your Environment

- Generator Model: 64620
- Generator Registers: Done
- Genmon Version: 1.13.33

Any chance of a pointer to get the snmp working please?

jgyates, 2020-09-29:
I have not see this issue before. After some research I believe this issue is related to this thread:

https://stackoverflow.com/questions/42514292/pysnmp-typeerror-setcomponentbytype-got-multiple-values-for-keyword-argumen

Genmon uses both of these libraries (pyasn1 and pysnmp). The exception throw is internal to pysnmp as genmon does not make a call to setComponentByType but I assume that pysnmp does. We only trap the unhandled exception.

If this is the case you have two choices:
1) Leave things like they are. 
2) Upgrade your pip libraries that genmon uses with this command, from the genmon directory:
       
        ./genmonmaint.sh -r

Regarding the error in your genloader log, (Error in InstallLibrary using pip ) this is due to the command is failing on your system:

     sudo pip install pysnmp

This can be caused by a few things, here are a few suggestions:

1) No active internet connection while attempting to install
2) The pip repository was down while attempting to install 
3) There is some type of corruption in your pip index files

Thanks for reporting this. It is good to get data points on what is working or not working from people, but I do think that this is something unique to your system. If you want to troubleshoot this we can start with the genmonmaint command listed above and then see if you can run "sudo pip install pysnmp".

jgyates, 2020-10-01:
I am closing this issue for now. If you have any questions on this topic feel free to post them to this thread.

closed 2020-10-01 by jgyates