jgyates/genmon #445: Software improvement / multiple instances of Genmon / genmqtt.py and genmqtt.conf update

liltux, 2020-10-02:
@jgyates  I am working on a multiple generator Genmon system and I found a problem with the genmqtt.py and .conf files.  It appears that in the genmqtt.py, when the mqtt instance is initialized it is using client id, "genmon".  If I have multiple instances of the genmqtt.py running, now all of them are the same client id.  Can this be changed to a variable set on the Add-ons page rather than hard coded.  I would submit a pull request, but last time i did one of those, we know how that went.
Thanks! ; )

jgyates, 2020-10-02:
That is easy enough to do. I will check that in for the next release.

jgyates, 2020-10-03:
This changed is included in V1.15.0 which is now checked in.

closed 2020-10-03 by jgyates

liltux, 2020-10-05:
@jgyates fyi, works perfect.

jgyates, 2020-11-16:
NOTE: A feature has been added in the upcoming release (V1.15.11) that affects running multiple instances of genmon. To run multiple instances of genmon (or any add on) you must set the multi-instance setting to True in the genmon.conf file, otherwise loading multiple instances will fail. More details are here:

https://github.com/jgyates/genmon/wiki/Appendix-J-Multiple-Instances-of-Genmon