jgyates/genmon #446: Unable to identify Molex connection in generator

DJT1423, 2020-10-05:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I have configured the Pi and have my cable but I have been unable to identify a free 8 pin Molex connector in the generator that it is supposed to attached to. See photos. Can you point me in the right direction? Thoughts, suggestions?

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

See photos

### Your Environment
[image]
[image]
[image]
[image]
[image]
[image]

- Generator Model: G0070310, 11KW
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

JustApu, 2020-10-05:
Look behind the "Accessory Port" stickers.

jgyates, 2020-10-05:
As @JustApu  mentioned it should be behind Accessory Port sticker. Port 1 if memory serves. These threads have some pictures:

See this thread:
https://github.com/jgyates/genmon/issues/85

or this one:
https://github.com/jgyates/genmon/issues/274

jgyates, 2020-10-05:
FYI you can search the issues on keywords. Just remove the search phrase "is:open" from the search window and add something like "Accessory"

DJT1423, 2020-10-05:
Awesome. Thanks for the quick help (and the search instruction!). David

closed 2020-10-05 by DJT1423