jgyates/genmon #447: External Switch Utility Power not reporting correctly

bbrown-sc, 2020-10-05:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

Correctly representing External Switch Utility Power

### Actual Behavior

Switch Utility Power dashboard meter bouncing back and forth between "0 kW" and "36280 kW"

### Steps to Reproduce (including precondition)

Observing Switch Utility Power dashboard meter

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: 2008 generac propane 25kW industrial liquid cooled QT02524 -  H-100 control panel - 400A HTS "W" Type transfer switch
- Generator Registers: Submit via the About page
- Genmon Version: V1.15.00

jgyates, 2020-10-05:
You submitted your registers however there appears to be something wrong with your email as I keep getting the same register dump every 2 minutes.

Submit your Logs (not registers) on the about page and wait 30 seconds and restart your pi.

jgyates, 2020-10-05:
Based on the limited data I have, you genmon installation is corrupt. 

Your email is repeatedly sending the same email over and over
It is sending only a partial email when you send the registers (half of the data is missing)

bbrown-sc, 2020-10-05:
Sorry about that. Let me reboot and submit the logs.

bbrown-sc, 2020-10-05:
Looks like the cause of the repeated (and partial) registers/logs emails is due to the following:

_Error SMTP sendmail: (500, 'Line length exceeded. See RFC 2821 #4.5.3.1.') : mymail.py:441_

I didn't make any changes to sendmail from stock Raspberry Pi OS Desktop install. I only configured mymail.conf.

jgyates, 2020-10-05:
No worries. I got your logs. Looks like the email issues was your mail server is strictly enforcing RFC 2821 #4.5.3.1. which requires limits on long lines in email bodies. I have not seen that before. I checked in a change in V1.15.01 that should fix this.

On to your original issue. Is your utility power showing up correctly in the web interface? I am trying to determine what the value should be (at least in the ball park). 

The most likely cause of a modbu value changing unexpectedly would be due to a modbus timeout. In some situations when a timeout occurs, the receiving end (the controller in this case) will respond with a packet, but the sending (genmon in this case) has already moved on and sent a request for the next packet. This is a limitation of the modbus protocol. If you are in a situation where some timeouts are expected (for example TCP over wireless), one thing you can do is add a few seconds to the modbus timeout duration. You can do this on the Advanced Settings page. Double click on the gear icon in the upper right of the web interface and you will see a setting named "Additional Modbus Timeout (sec)". Make this value a 3 and click save at the bottom.

jgyates, 2020-10-05:
Once you upgrade to V1.15.01 try sending your logs or registers again.

jgyates, 2020-10-05:
Got it. Works. Let me know if the modbus timeout improves your problem.

bbrown-sc, 2020-10-05:
I'm not sure there's even a "External Switch Utility Power" register/value. I don't see one on the H-100 control panel or the genlink sw utility. Where do you get the information explaining the registers? Did you reverse engineer the registers or is this documented in a service manual somewhere?

jgyates, 2020-10-05:
That is possible. Can anyone else chime in who is using a H-100 with the "HTS/MTS/STS Transfer Switch" option enabled?One possibility is that the utility power reading is only available on some of the HTS/MTS/STS switches.

jgyates, 2020-10-06:
Looks like we can replicate this with genlink. Closing for now, feel free to comment here if you have additional data.

closed 2020-10-06 by jgyates